mavlink/MAVSDK-Java #135: Create geofence that uses a circle instead of a polygon

mfran89, 2023-07-24:
Hey team! 

I am currently looking into passing a keepout geofence circle using the geofence functionality through the System class (system.getGeofence()). So far I am able to pass polygons to be used as geo fences (both inclusive and exclusive). However I was hoping there was a way to pass a center point (lat, long) and radius , like a Drawingcircle object, as a geofence but I do not see that as an ability. For reference I am trying to pass something like this mavlink command https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_FENCE_CIRCLE_EXCLUSION , does the java mavsdk allow the end user to pass this somehow? 

Currently the only option I can think of is creating a polygon in the shape of a circle mathematically but I am not sure if there are other ways to do this either.

JonasVautherin, 2023-07-24:
Hmm I think it is supported in the `main` branch of MAVSDK, which hasn't been released (it will be MAVSDK v2). See here: https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/geofence/geofence.proto#L44

Until then, I'm afraid you have to build MAVSDK from sources.

mfran89, 2023-07-24:
So for now if I dont build from source would you suggest that creating a polygon based of the center and radius of the circle is my best bet, and if so should I worry about a max set of points for the polygon or anything like that?

JonasVautherin, 2023-07-24:
I think that in MAVLink, the geofence uses the mission protocol, which is extremely inefficient (it requires one round-trip per item). So yeah, if you don't want to compiles the `main` branch, then use as few points as you can accept :sweat_smile:.

mfran89, 2023-07-24:
Just making sure I understand this but there is no way other way to pass the mavlink command I mentioned about correct? There is no method of Mavlink Passthrough for mavsdk java correct?

JonasVautherin, 2023-07-24:
> There is no method of Mavlink Passthrough for mavsdk java correct?

Hmm maybe by hijacking the mission_raw plugin? Not sure though. I would just compile from sources.

> Also when will MavSDK v2 be released?

It is not planned, but the roadmap to v2 is here: https://github.com/mavlink/MAVSDK/issues/2002

mfran89, 2023-07-25:
Hypothetically if I did want to build from source would I  just need to pull this repo and update the proto files for this repo to use the main branch of MAVSDK-Proto, and then build it ? Or would I need to also to some other steps to get this properly built from source?

JonasVautherin, 2023-07-25:
You need to build https://github.com/mavlink/MAVSDK for your platform (that's where `mavsdk_server` comes from), build this repo, and use them together. Are you building for Android?

mfran89, 2023-07-25:
Yes I am building for android, will go through the steps you mentioned.

JonasVautherin, 2023-07-25:
I would say that the first step is to cross-compile MAVSDK for Android. Check how we do it in CI: https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L385

closed 2023-07-28 by mfran89