mavlink/MAVSDK-Java #136: Mavsdk not giving all data when connected to drone

RattanIndia-NeoSky, 2023-08-07:
i have implemented mavsdk java in my code to connect with drone. When I connect drone with my app in telemetry only few function gives me data i.e. telemetry.health(), telemetry.statusText(), Telemetry.isArmed()  The rest is not giving any data . But once i connect the telemetry with qgc or mission planner and than i connect the drone to my it start giving all data until i restart my drone. 

Don't know why this happening..

Please reply...

JonasVautherin, 2023-08-10:
Should we do that in MAVSDK? Like in the telemetry subscribe functions, do `if (ardupilot) { set_rate_... }` maybe? :thinking:

julianoes, 2023-08-11:
Yer, not sure.

akirahrkw, 2023-08-15:
hi
I had same issue recently with telemetry.heading.

Do I need to call setRateXXXX method?
in that case, do you have any ideas which setter method I should call for heading, as it seems there is no setRateHeading.

julianoes, 2023-08-16:
That's tricky. For that you have to have a look in the internal to check which message it depends on and which message therefore needs to be requested.

akirahrkw, 2023-08-17:
@julianoes Thank you for the reply! I think I got to know why

following your advice, I took a look at the code of MAVSDK for the implementation, and qgroundcontrol to see how QGC handle the data for compass heading.

seeing telemetry_impl.cpp on MAVSDK,  telemetry.heading is based on GPS data and seeing Vehicle.cc on QGC, compass heading uses yaw (attitudeEuler) which I wanted. 
I think as I was testing my app indoor, I didn't receive the stream of GPS...
As I was just misunderstanding the definition, my case perhaps is not relating to setRateXXXX.

Thanks!

julianoes, 2023-08-18:
Oh, heading is using the GPS message? That's indeed tricky.

akirahrkw, 2023-08-18:
I just quickly checked the implementation of MAVSDK (so my understanding might be wrong) ,

in telemetry_impl.cpp
```
void TelemetryImpl::process_global_position_int(const mavlink_message_t& message)
{
...
...

    {
        Telemetry::Heading heading;
        heading.heading_deg = (global_position_int.hdg != std::numeric_limits<uint16_t>::max()) ?
                                  static_cast<double>(global_position_int.hdg) * 1e-2 :
                                  static_cast<double>(NAN);
        set_heading(heading);
    }
...
...

```

seeing the part above, telemetry heading uses hdg on this message
https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT

julianoes, 2023-08-20:
Ok, from GLOBAL_POSITION_INT, that makes sense. So you can just set the rate of that message. But you're problem is that you don't have a global position and hence don't see heading?

akirahrkw, 2023-08-21:
yes that was my problem (I'm using yaw of attitudeEuler)

mfran89, 2023-08-21:
I am not seeing any data come out when I use the following subscription 

```
_disposables.add(drone.getTelemetry().getStatusText().subscribe(statusText
                -> Log.d(TAG, "status is " + statusText.getText()),
                error -> Log.d(TAG, "status error is " + error),
                () ->  Log.d(TAG,"Status Text subscription completed")));
```

I am able to get information for other telemetry information such as battery , flight mode, but nothing comes out for getStatusText. Do I need to use some type of setRate function to get the status text to return? FWIW I am using Ardupilot pixhawk , along with mavsdk server.

mfran89, 2023-08-22:
Ok initially I thought the message status text would just be published at a certain rate with out a command needing to be sent. However that is not the case and I am not sure how to send the request for the status text through the mavsdk. Are there any examples that I could use to send a status text request (https://mavlink.io/en/messages/common.html#STATUSTEXT)?

julianoes, 2023-08-23:
Statustexts are usually sent out on startup or error, at least by ArduPilot: e.g. when a check fails before arming.

julianoes, 2023-08-23:
Ok, I'm closing this issue now, assuming all questions are answered.

closed 2023-08-23 by julianoes