mavlink/MAVSDK-Java #137: Android example

Zxr1230, 2023-08-14:
I am trying to use the Android example to control my ardupilot drone. I installed the Android example app in the Herelink ground controller. I wonder how I should connect the app to the drone. Thank you!

JonasVautherin, 2023-08-14:
MAVSDK, by default, listens for MAVLink on udp/14540. This is because we started with PX4, and that is what it does. I believe, however, that Ardupilot works differently, right? How does the MAVLink connection work with Ardupilot? Does MAVSDK have to start sending MAVLink heartbeats to the drone? Does the drone broadcast heartbeats?

Zxr1230, 2023-08-14:
> MAVSDK, by default, listens for MAVLink on udp/14540. This is because we started with PX4, and that is what it does. I believe, however, that Ardupilot works differently, right? How does the MAVLink [...]

Thanks for replying! I think it's pretty much the same with the PX4. It does broadcast heartbeats. I am still learning. Does the example app work with the PX4 gazebo simulation? https://docs.px4.io/main/en/sim_gazebo_gz/

JonasVautherin, 2023-08-15:
Maybe you could start with https://github.com/JonasVautherin/px4-gazebo-headless/. That's a PX4 SITL, but it would confirm that your example app works before moving on with the Ardupilot setup.

Zxr1230, 2023-08-16:
> Maybe you could start with https://github.com/JonasVautherin/px4-gazebo-headless/. That's a PX4 SITL, but it would confirm that your example app works before moving on with the Ardupilot setup.

Thanks! I also wonder when I am using this [android-client](https://github.com/mavlink/MAVSDK-Java/tree/main/examples/android-client) app, do I need to install the mavsdk_server? Or the app comes with it and I just need to click the "RUN MAVSDK SERVER" button in the app.

Zxr1230, 2023-08-17:
> The android client examples installs mavsdk_server here: https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/app/build.gradle#L64
> 
> ```
>     implementation 'io.mavsdk:mavsdk-server:1.2.1'
> ```
> 
> And it runs it here: https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/app/src/main/java/io/mavsdk/androidclient/MapsActivity.java#L251

Thank you! So I don't need to install anything manually on the Android device right? I just need to click this button right?
[image]

closed 2023-11-27 by Zxr1230