mavlink/MAVSDK-Java #138: mavsdk server hangs with the run function

mfran89, 2023-08-15:
Currently using 'io.mavsdk:mavsdk-server:1.3.2'

My current architecture uses mavsdk server with a tcp connection , and I am able to generally connect to the Ardupilot device. However there are scenarios in which the mavsdkserver obj will hang when I make a call to _mavsdkServer.run(systemAddress). The system address being tcp://127.0.0.1:8787. Looking at the log I see that mavsdk will return 
```
connect error: Connection refused
Connection failed: Socket connection error
```

I am trying to understand why these errors could show up and if others have seen this same one.  Also is there a way to have mavsdk server return instead of hang when it cannot find a connection?

JonasVautherin, 2023-08-15:
> The system address being tcp://127.0.0.1:8787

This means that Ardupilot runs on the same machine as MAVSDK. Is that correct? It seems like it, since you mention that it sometimes work.

It will fail if Ardupilot is not running, because MAVSDK won't be able to initiate a connection. Also it may fail if there is already a connection to Ardupilot, if Ardupilot does not support multiple connections (does it? I don't know).

What do you think?

mfran89, 2023-08-17:
Apologies I thought I wrote back yesterday. Mavsdk is not running on the same machine as Ardupilot. Mavsdk runs on android tablet, and Ardupilot runs on a pixhawk. Is there an example I can reference to make sure I am going to through the right steps to use a static member mavsdk server to create a connection to the AP, and then when done disconnect. Currently I do the following:

```
//connect
_mavsdkServer = new MavsdkServer();
 int mavsdkServerPort = _mavsdkServer.run(systemAddress);
_ghdrone = new System(MAVSDK_SERVER_IP, mavsdkServerPort);
//disconnect
_ghdrone.dispose();
_mavsdkServer.stop();
_mavsdkServer.destroy();
```

mfran89, 2023-08-21:
issue resolved, it looks like I wasnt actually creating a new instance of the mavsdk server like I pointed out before before calling the run (), which means I need to create a new instance of the server each time I disconnect and reconnect . So the above code I used will handle preventing the hanging issue I mentioned above.

closed 2023-08-21 by mfran89