mavlink/MAVSDK-Java #143: Unable to get flight mode

Zxr1230, 2023-12-08:
I am trying to run the Mavsdk in the android-client example.  The Android example app is installed in the Herelink ground controller. After connecting the drone, the app displays the flight mode unknown but the flight mode displayed by the drone is LOIT. 
Thanks in advance! 
[image]
[image]

JonasVautherin, 2023-12-08:
Somehow I feel like I have seen that picture already. Did you open another issue somewhere else?

JonasVautherin, 2023-12-08:
Does the rest work? E.g. do you get correct battery updates?

Zxr1230, 2023-12-09:
> Does the rest work? E.g. do you get correct battery updates?

Thanks! I think it works. But when I press the take-off button, the app will crash. 
[image]

JonasVautherin, 2023-12-09:
> But when I press the take-off button, the app will crash.

I guess it throws an exception. What does it say?

Zxr1230, 2024-01-12:
> > But when I press the take-off button, the app will crash.
> 
> I guess it throws an exception. What does it say?

Sorry for the late reply. I was working on something else. It doesn't throw any exception. The app just crashed and quit. Thank you!
[image]

JonasVautherin, 2024-01-12:
Well the app probably crashes by throwing an exception, but you won't see that in the Android popup :blush:. Did you check logcat?

Zxr1230, 2024-01-12:
> Well the app probably crashes by throwing an exception, but you won't see that in the Android popup 😊. Did you check logcat?

Thank you! It said the command was denied. It means the drone denied the command right?
[image]
[image]

JonasVautherin, 2024-01-12:
> It means the drone denied the command right?

Yes!

Zxr1230, 2024-01-12:
> > It means the drone denied the command right?
> 
> Yes!

Thank you! What could be the issue that makes flight mode doesn't work? Can this app control the drone to fly directly? For example fly it higher or lower, forward or backward.

mfran89, 2024-08-15:
I would also like to jump on this thread, I currently am also having issues accessing the flight mode via 
```
_disposables.add(drone.getTelemetry().getFlightMode().subscribe(flightMode
                -> {
            Log.d(TAG,"flight mode:" + flightMode.name() + "flight mode number " + flightMode.ordinal());
        }));
```

I am able to get position telemetry and other variables but for some reason flight mode returns 0. Where as on mission planner I see the right flight mode. I am currently using Ardupilot and mavsdk 1.3.1 and mavsdk server 1.3.2.

JonasVautherin, 2024-08-15:
> I am currently using Ardupilot and mavsdk 1.3.1 and mavsdk server 1.3.2.

I don't remember exactly, but some Ardupilot features were added for MAVSDK v2. Flight mode may be one?

closed 2024-08-20 by JonasVautherin