mavlink/MAVSDK-Java #146: Update MavSDK to 2.1.0

tkocik92, 2024-02-07:
Is it possible to update `MAVSDK-Java` to use `MavSDK 2.1.0`?

JonasVautherin, 2024-02-07:
Good point, let me see that ASAP.

JonasVautherin, 2024-02-07:
I just released mavsdk-java v2.0.0 to the maven repository. The packages should be there in a few hours. Would you mind trying them and letting me know if they work for you?

JonasVautherin, 2024-02-07:
It is already there: https://central.sonatype.com/artifact/io.mavsdk/mavsdk

tkocik92, 2024-02-08:
I confirmed the updates are available, but `implementation 'io.mavsdk:mavsdk:2.0.0'` appears to be missing its classes? `import io.mavsdk.System` and all related classes are throwing errors:
[image]

JonasVautherin, 2024-02-08:
You're right, apparently with the newer version of gradle this got dropped in the publish task. I pushed again (io.mavsdk:mavsdk:2.0.1), would you mind trying that?

tkocik92, 2024-02-09:
Everything is working again. Thanks!

closed 2024-02-09 by tkocik92