mavlink/MAVSDK-Java #149: Multiple systems on the network

tkocik92, 2024-02-09:
MAVSDK-Java Version: `2.0.1`
MavSDK-Server Version: `2.0.0`

I am running into an issue when there are multiple Systems detected on a network. My implementation looks like the following:

```
val mavSdkPort = mavSdkServer.run("udp://$ipAddress:$port")
val system = System(ipAddress, mavSdkPort)
```
Where `ipAddress = "0.0.0.0"` and `port` is the port of the system I am trying to connect to.

If only 1 system is on the network, I am able to connect perfectly fine. If 2 systems are on the network, I am connected to both Systems with no way to discern between the two. Logcat looks like the following:

```
Mavsdk           MAVSDK version: v2.1.0
MAVSDK-Server    Running mavsdk_server with connection url: udp://0.0.0.0:14550
Mavsdk           Waiting to discover system on udp://0.0.0.0:14550...
Mavsdk           New system on: 192.168.4.223:43620 (with system ID: 1) // Simulator
Mavsdk           New system ID: 1 Comp ID: 1
Mavsdk           Component Autopilot (1) added.
Mavsdk           Component Gimbal (154) added.
Mavsdk           New system on: 10.X.X.X:14550 (with system ID: 1) // System I am attempting to connect to
```

If I change `ipAddress` (mentioned above) to the IP of the desired system, MavSDK Server never finishes booting up. What can I do differently?

JonasVautherin, 2024-02-09:
The problem is that MAVSDK-Java expects exactly one system on the port it listens to (here udp://:14550). It is the same with all the language bindings (e.g. MAVSDK-Swift, MAVSDK-Python). What you have to do is demultiplex the MAVLink stream, and send it to different instances of mavsdk_server.

Say you have two drones, both broadcasting on udp://:14550. You can run a mavlink-router that listens on 14550 and forwards to 14551 and 14552. Then you can run on mavsdk_server with `mavSdkServer.run("udp://:14551")` and the other with `mavSdkServer.run("udp://:14552")`.

Does that make sense?

ProgrammedForAdventure, 2024-11-05:
I'm experiencing this same issue when trying to deconflict multiple drones on the same network, such as when there is a residual drone still sending data to my Java instance's IP address. Is there any hope on the horizon to be able to filter by System ID, or to even get the System ID from any received messages?

JonasVautherin, 2024-11-06:
If they have a different system id, then it should not conflict :thinking:: MAVSDK-Java will just connect to the first one it encounters, right?

My last comment is still valid, though: you can demultiplex based on the source IP :+1:.

ProgrammedForAdventure, 2024-11-06:
They conflict when it comes to receiving `System` flowable subscription updates, such as these (`drone` is a `System` object):
```
drone?.core?.connectionState?.distinctUntilChanged()?.subscribe({ value ->
    // Begins receiving updates for both System connections
}
```

When I start the new connection via a new `System`, if there is an existing `System`, I make sure to destroy it, stop the server, and clean up disposables before creating a new one:
```
disposables.forEach { it.dispose() }
disposables.clear()
drone?.dispose()
drone = null
mavSdkServer.stop()
mavSdkServer.destroy()
```

And then I create a new `System`:
```
val mavSdkPort = mavSdkServer.run("${protocol.type}://$ipAddress:$port")
Log.d(TAG, "Creating System on port $mavSdkPort, ${protocol.type}://$ipAddress:$port")
drone = System(ipAddress, mavSdkPort)
```

I only have problems when a second drone is up and broadcasting to my device's IP address on port `14550`. Here are some logs I have. Thanks for getting back so quickly!

```
---------------------------- PROCESS STARTED (26361) for package com.example.app ----------------------------
2024-11-05 10:29:21.210 26361-26419 Mavsdk                  com.example.app                I  MAVSDK version: v2.12.2
2024-11-05 10:29:21.210 26361-26419 MAVSDK-Server           com.example.app                D  Running mavsdk_server with connection url: udp://0.0.0.0:14550
2024-11-05 10:29:21.210 26361-26419 Mavsdk                  com.example.app                I  Waiting to discover system on udp://0.0.0.0:14550...
2024-11-05 10:29:41.306 26361-26422 Mavsdk                  com.example.app                I  New system on: <IP_ADDRESS_1>:41279 (with system ID: 1)
2024-11-05 10:29:41.306 26361-26422 Mavsdk                  com.example.app                D  New system ID: 1 Comp ID: 1
2024-11-05 10:29:41.307 26361-26422 Mavsdk                  com.example.app                D  Component Autopilot (1) added.
2024-11-05 10:29:42.138 26361-26422 Mavsdk                  com.example.app                W  Vehicle type changed (new type: 13, old type: 0)
2024-11-05 10:29:42.138 26361-26422 Mavsdk                  com.example.app                D  Discovered 1 component(s)
2024-11-05 10:29:42.140 26361-26421 Mavsdk                  com.example.app                I  System discovered
2024-11-05 10:29:42.149 26361-26419 Mavsdk                  com.example.app                I  Server started
2024-11-05 10:29:42.149 26361-26419 Mavsdk                  com.example.app                I  Server set to listen on 0.0.0.0:46751
2024-11-05 10:29:42.149 26361-26419 MAVSDK-Server           com.example.app                D  mavsdk_server is now running, listening on port 46751
2024-11-05 10:29:42.251 26361-26422 Mavsdk                  com.example.app                D  Component Gimbal (154) added.
2024-11-05 10:29:42.376 26361-26422 Mavsdk                  com.example.app                D  Using gimbal protocol v2
2024-11-05 10:29:43.519 26361-26422 Mavsdk                  com.example.app                D  MAVLink: info: [logger] ./log/2024-11-05/16_29_40.ulg	
2024-11-05 10:29:43.559 26361-26420 Mavsdk                  com.example.app                W  sending again, retries to do: 5
2024-11-05 10:29:43.581 26361-26422 Mavsdk                  com.example.app                D  MAVLink: info: Failsafe mode activated	
2024-11-05 10:29:44.067 26361-26420 Mavsdk                  com.example.app                W  sending again, retries to do: 4

<------------- POWERED ON SECOND DRONE --------------------------->

2024-11-05 10:32:53.710 26361-26422 Mavsdk                  com.example.app                W  User callback queue too slow.
                                                                                                    See: https://mavsdk.mavlink.io/main/en/cpp/troubleshooting.html#user_callbacks
2024-11-05 10:32:55.731 26361-26422 Mavsdk                  com.example.app                W  User callback queue too slow.
                                                                                                    See: https://mavsdk.mavlink.io/main/en/cpp/troubleshooting.html#user_callbacks
2024-11-05 10:32:56.035 26361-26422 Mavsdk                  com.example.app                D  MAVLink: info: Manual control regained after 23.8s	
2024-11-05 10:32:56.994 26361-26419 Mavsdk                  com.example.app                I  MAVSDK version: v2.12.2
2024-11-05 10:32:56.995 26361-26419 MAVSDK-Server           com.example.app                D  Running mavsdk_server with connection url: udp://0.0.0.0:14550
2024-11-05 10:32:56.995 26361-26419 Mavsdk                  com.example.app                I  Waiting to discover system on udp://0.0.0.0:14550...
2024-11-05 10:32:57.000 26361-26895 Mavsdk                  com.example.app                I  New system on: <IP_ADDRESS_1>:41279 (with system ID: 1)
2024-11-05 10:32:57.000 26361-26895 Mavsdk                  com.example.app                D  New system ID: 1 Comp ID: 1
2024-11-05 10:32:57.000 26361-26895 Mavsdk                  com.example.app                D  Component Autopilot (1) added.
2024-11-05 10:32:57.023 26361-26895 Mavsdk                  com.example.app                D  Component Gimbal (154) added.
2024-11-05 10:32:57.066 26361-26895 Mavsdk                  com.example.app                I  New system on: <IP_ADDRESS_2>:14550 (with system ID: 77)
[... cut ...]
2024-11-05 10:32:59.834 26361-26895 Mavsdk                  com.example.app                D  Using gimbal protocol v2
2024-11-05 10:33:00.554 26361-26893 Mavsdk                  com.example.app                W  sending again, retries to do: 5
2024-11-05 10:33:00.665 26361-26890 MavHealthMonitor        com.example.app                E  handleHealthState: magnetNotOk
2024-11-05 10:33:01.059 26361-26893 Mavsdk                  com.example.app                W  sending again, retries to do: 4
```

ProgrammedForAdventure, 2024-11-06:
You are correct though about it latching onto the first connection it receives, thanks for pointing that out!

ProgrammedForAdventure, 2024-11-12:
@JonasVautherin Does MAVSDK-Java support passing IP addresses other than `0.0.0.0` in when creating the `mavsdk-server`? Such as `mavSdkServer.run("udp://192.168.1.50:14550")`? That way you are restricting it to listen over a specific IP?

JonasVautherin, 2024-11-12:
MAVSDK-Java passes the string to the C++ mavsdk_server. So if the underlying `mavsdk_server` supports the new `udpin`/`udpout` syntax, you could use that. I am just not sure when this was added... @julianoes do you know this from the top of your head?

julianoes, 2024-11-12:
This will come with MAVSDK v3, any time now.