mavlink/MAVSDK-Java #155: camera/possibleSettingOptions segfaults using MAVSDK-Java 2.0.1 and MAVSDK-server-Android 2.0.0

rayw-dronesense, 2024-03-13:
This did not occur with previous version 1.3.2

```
2024-03-13 17:39:40.622 25779-26537 libc++abi               com.dronesense.pilot                 E  terminating with uncaught exception of type std::invalid_argument: stoul: no conversion
2024-03-13 17:39:40.622 25779-26537 libc                    com.dronesense.pilot                 A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 26537 (RxCachedThreadS), pid 25779 (ronesense.pilot)
2024-03-13 17:39:40.633 25779-26540 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
2024-03-13 17:39:40.633 25779-26540 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
2024-03-13 17:39:40.633 25779-26540 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2024-03-13 17:39:40.633 25779-26540 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
2024-03-13 17:39:40.638 25779-26416 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2024-03-13 17:39:40.645 25779-26416 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
2024-03-13 17:39:40.725 25779-25779 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Landroid/view/View;->getViewRootImpl()Landroid/view/ViewRootImpl; (greylist, reflection, allowed)
2024-03-13 17:39:40.729 26548-26548 DEBUG                   pid-26548                            A  pid: 25779, tid: 26537, name: RxCachedThreadS  >>> com.dronesense.pilot <<<
2024-03-13 17:39:40.730 25779-25779 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden field Landroid/view/View$AttachInfo;->mVisibleInsets:Landroid/graphics/Rect; (greylist, reflection, allowed)
2024-03-13 17:39:40.730 25779-25779 ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden field Landroid/view/ViewRootImpl;->mAttachInfo:Landroid/view/View$AttachInfo; (greylist, reflection, allowed)
2024-03-13 17:39:40.730 26548-26548 DEBUG                   pid-26548                            A        #01 pc 00000000000d1cc0  /data/app/com.dronesense.pilot-RbUkTiXzrdIfRxYjrPikzQ==/base.apk!libc++_shared.so (offset 0xfb3000) (BuildId: a59088f9640cd272bc9542d94dc84a0c88afd558)
```

JonasVautherin, 2024-03-13:
Maybe you could try to reproduce it in C++? That would make it easier to debug.

rayw-dronesense, 2024-03-14:
@JonasVautherin Thanks for getting back to me, I'm not sure how one would do that on Android.

I was able to get more detailed logs by removing log filters,

```
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and float
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and float
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and float
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and float
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and float
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and uint32_t
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Comparison type mismatch between uint8_t and uint32_t
2024-03-14 11:55:06.930  4661-5594  Mavsdk                  com.dronesense.pilot                 W  Trying to compare different types.
--------- beginning of crash
2024-03-14 11:55:06.930  4661-5594  libc++abi               com.dronesense.pilot                 E  terminating with uncaught exception of type std::invalid_argument: stoul: no conversion
2024-03-14 11:55:06.930  4661-5594  libc                    com.dronesense.pilot                 A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 5594 (RxCachedThreadS), pid 4661 (ronesense.pilot)
2024-03-14 11:55:06.932  4661-5598  ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
2024-03-14 11:55:06.957  4661-5407  ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
2024-03-14 11:55:06.957  4661-5598  ronesense.pilo          com.dronesense.pilot                 W  Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
...
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  Build fingerprint: 'qti/kona/kona:10/V1.0.0.17/eng.jenkin.20230915.165135:userdebug/test-keys'
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  Revision: '0'
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  ABI: 'arm64'
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  Timestamp: 2024-03-14 08:55:07-0700
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  pid: 4661, tid: 5594, name: RxCachedThreadS  >>> com.dronesense.pilot <<<
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  uid: 10120
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A  Abort message: 'terminating with uncaught exception of type std::invalid_argument: stoul: no conversion'
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A      x0  0000000000000000  x1  00000000000015da  x2  0000000000000006  x3  0000007d6f4086e0
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A      x4  ff6d6e6872716475  x5  ff6d6e6872716475  x6  ff6d6e6872716475  x7  7f7f7f7f7f7f7f7f
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A      x8  00000000000000f0  x9  cc32b44918147016  x10 0000000000000001  x11 0000000000000000
2024-03-14 11:55:07.073  5614-5614  DEBUG                   pid-5614                             A      x12 fffffff0fffffbdf  x13 0000000065f31dda  x14 003731a364064800  x15 00004350de94e70b
[... cut ...]
2024-03-14 11:55:07.074  5614-5614  DEBUG                   pid-5614                             A        #00 pc 00000000000830f0  /apex/com.android.runtime/lib64/bionic/libc.so (abort+160) (BuildId: 21847aa9757f000b0461310a9f5e6e51)
2024-03-14 11:55:07.074  5614-5614  DEBUG                   pid-5614                             A        #01 pc 00000000000d1cc0  /data/app/com.dronesense.pilot-GkDbKAZ1krjjpU4_zdAUxg==/base.apk!libc++_shared.so (offset 0xfb3000) (BuildId: a59088f9640cd272bc9542d94dc84a0c88afd558)
```

rayw-dronesense, 2024-03-14:
I was finally able to grab the Tombstone for the crash,

[tombstone_33.zip](https://github.com/mavlink/MAVSDK-Java/files/14605059/tombstone_33.zip)

JonasVautherin, 2024-03-14:
>  I'm not sure how one would do that on Android.

No need to do it from Android: try to reproduce in C++ from your computer :blush:

rayw-dronesense, 2024-03-14:
@JonasVautherin - Hmm...I'm unsure on how to do this on the computer - the camera module only emits with the real drone, as far as I'm aware, and it is connected via this Android-based controller.

JonasVautherin, 2024-03-14:
Oh, I see... What controller is that?

julianoes, 2024-03-15:
```
std::invalid_argument: stoul: no conversion'
```

Can you share the xml camera definition file that it parses? Using that we should be able to spot the problem or reproduce it?

julianoes, 2024-03-18:
This doesn't look right:

```
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
03-14 08:59:28.011  5974  6729 W Mavsdk  : Comparison type mismatch between uint8_t and float
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
03-14 08:59:28.011  5974  6729 W Mavsdk  : Comparison type mismatch between uint8_t and float
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
03-14 08:59:28.011  5974  6729 W Mavsdk  : Comparison type mismatch between uint8_t and float
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
03-14 08:59:28.011  5974  6729 W Mavsdk  : Comparison type mismatch between uint8_t and uint32_t
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
03-14 08:59:28.011  5974  6729 W Mavsdk  : Comparison type mismatch between uint8_t and uint32_t
03-14 08:59:28.011  5974  6729 W Mavsdk  : Trying to compare different types.
```

Given that the xml looks correct, presumably the type set in param_get_ext or param_set_ext is not correctly set by the camera implementation. Can you check that part?

rayw-dronesense, 2024-03-18:
Hi @julianoes ,

Unfortunately we don't have access to the source code for the drone.

What we do know is, the same logic and drone works using the mavsdk-server-android 1.3.2

julianoes, 2024-03-18:
Right, so it could also be a mavsdk regression then... :thinking:

I guess we need to add a bunch of debug output and then try that.

rayw-dronesense, 2024-03-19:
@julianoes - Nope, this was repro'd with the latest artifacts published on maven.

julianoes, 2024-03-19:
Right, well, I need to put this on my todo list. If it's urgent to you, consider [github sponsoring](https://github.com/sponsors/julianoes).

rayw-dronesense, 2024-11-22:
@julianoes / @JonasVautherin - Found the root cause of this issue.  The underlying logic for parsing and comparing ParamValues is expecting a string to be numeric, using functions like stoul to check equality. However, one code path was erroneously passing a string representation of ParamValue class instead of the numeric value.

The fix can be done in

```
bool CameraImpl::get_possible_options(
    const std::string& setting_id, std::vector<Camera::Option>& options)
```

whereby,

```
        std::stringstream ss{};
        ss << value;
```

needs to be changed to

```
        std::stringstream ss{};
        ss << value.get_string();
```

JonasVautherin, 2024-11-23:
Would you mind opening a PR with this fix?

rayw-dronesense, 2024-12-04:
Part of 2.1.4 release

closed 2024-12-04 by rayw-dronesense