mavlink/MAVSDK-Java #157: Receive no callback for System.calibration.calibrateAccelerometer(). I'm using MavSdk v2.0.1, MavSdk Server v2.0.0, and Ardupilot v4.4.4

Ba0Nguyen, 2024-05-09:
val currentSystem = System(MainApplication.BACKEND_IP_ADDRESS, port)

currentSystem.calibration.calibrateMagnetometer().subscribe(
{
    Log.i("calibrate", "Process: ${it.statusText} ${it.progress}")
}, 
{e ->
    Log.d("calibrate", "ERROR: ${e.cause} ${e.message}")
})

JonasVautherin, 2024-05-09:
Did you try the examples, e.g. https://github.com/mavlink/MAVSDK-Java/blob/main/examples/java-client/src/main/java/io/mavsdk/example/Calibrate.java?

Also, do you receive MAVLink messages at all? E.g. you could try receiving the battery updates or sending an arm command. Just to confirm that MAVLink goes through (maybe something is wrong with the calibration for Ardupilot, I only used it with PX4).

Ba0Nguyen, 2024-05-09:
I am working with Ardupilot v4.4.4. I tested the code you sent and receive no response. I did try sending the command twice, the first time I received "Starting Gyro calibration" since it is subscribed successfully, the second time I received a response of "Busy" which indicates the calibration is started. 
 
 I also received MavLink messages with arm() and Battery updates and other telemetry data.

julianoes, 2024-05-09:
Ok, not sure if calibration is compatible with ArduPilot. This likely requires some work.

JonasVautherin, 2024-05-09:
@Ba0Nguyen: do you know how the calibration works with Ardupilot? Is it specified somewhere? Would be nice to compare to PX4 and see what is different.

Ba0Nguyen, 2024-05-09:
https://mavlink.io/en/messages/ardupilotmega.html   MAV_CMD_ACCELCAL_VEHICLE_POS
From attached link, I think Ardupilot might send a message to tell the GCS what position to put the drone in, or the GCS can send a command to Ardupilot what position the drone is currently in. I'm not sure if it is possible to choose which approach to set for Ardupilot.

closed 2024-05-12 by Ba0Nguyen

JonasVautherin, 2024-05-12:
Did you find a solution?

Ba0Nguyen, 2024-05-12:
I believe for Ardupilot, I have to place the aircraft and a certain position and send a MavCommmand to the flight controller, to which it might send me back a ProgressData response. It still just a theory, but I am adding a feature to the Java-SDK and Core to support sending the MavCommand message.