mavlink/MAVSDK-Java #161: How to use mavsdk_server_manylinux2010-x64?

wangpuxia, 2024-06-03:
I download mavsdk_server_manylinux2010-x64 file,but how to runing it?

JonasVautherin, 2024-06-03:
Did you try `./mavsdk_server_manylinux2010-x64`?

wangpuxia, 2024-06-03:
-bash: .: mavsdk_server_manylinux2010-x64: cannot execute binary file

divyanshupundir, 2024-06-03:
Please try this:

```bash
chmod +x mavsdk_server_manylinux2010-x64
```

After this you should be able to run the binary as @JonasVautherin has explained earlier.

julianoes, 2024-06-03:
@wangpuxia what machine is this on? Linux, macOS, x64, arm?

wangpuxia, 2024-06-03:
> @wangpuxia what machine is this on? Linux, macOS, x64, arm?

CentOS Linux release 7.7

wangpuxia, 2024-06-03:
> Please try this:
> 
> ```shell
> chmod +x mavsdk_server_manylinux2010-x64
> ```
> 
> After this you should be able to run the binary as @JonasVautherin has explained earlier

i tried,but error is same

wangpuxia, 2024-06-03:
[image]

divyanshupundir, 2024-06-03:
There should be a `/` between the `.` and the binary name with no white space.

```bash
./mavsdk_server_manylinux2010-x64
```

wangpuxia, 2024-06-03:
> There should be a `/` between the `.` and the binary name with no white space.
> 
> ```shell
> ./mavsdk_server_manylinux2010-x64
> ```

really help me ,thanks a lot

wangpuxia, 2024-06-03:
[image]
runing mavlink-server succful on a server computer(with public IP), and open a port 18888

but at other computer,I cant connect this port 18888, telnet failed ,nmap failed
[image]

julianoes, 2024-06-03:
You're trying nmap on TCP when it's probably UDP.

julianoes, 2024-06-03:
Nevermind, it's TCP. In this case you need to look at your setup and firewall settings. It looks like your mavsdk_server is running in a virtual machine? Why are you running it in a virtual machine but then connecting outside in Windows?

closed 2024-06-04 by wangpuxia