mavlink/MAVSDK-Java #164: how to catch the simulator warning message

wangpuxia, 2024-06-11:
code like this:

drone.getMission ( ).startMission ( ).doOnError ( err -> logger.info ( " mission start exception : " + err.getMessage ( ) ) ).andThen ( drone.getMission ( ).getMissionProgress ( ).doOnError ( err -> logger.info ( " mission start progress exception : " + err.getMessage ( ) ) ).map ( mission ->
					{
						return mission.getCurrent ( ).doubleValue ( ) / mission.getTotal ( ).doubleValue ( ) ;

					} ).distinctUntilChanged ( ) ).subscribe ( percent ->
					{
						logger.info (String.valueOf ( percent ) ) ;
					} , throwable ->
					{
						logger.error (  throwable.getMessage ( ) ) ;
					} ) ;

jmavsim simulator console warning : [mission feasibility_checker] First waypoint too far away: 1208m, 900 max

but logger is :mission start exception : UNKNOWN: Denied

how to catch the jmavsim warning message ?

JonasVautherin, 2024-06-11:
I don't understand your question. The error is:

> First waypoint too far away: 1208m, 900 max

If you set a first waypoint closer to the location of the drone, then `startMission()` should not return an error.

julianoes, 2024-06-11:
Check parameter: https://docs.px4.io/v1.14/en/advanced_config/parameter_reference.html#MIS_DIST_1WP

closed 2024-06-11 by julianoes