mavlink/MAVSDK-Java #166: I get `TIMEOUT: Timeout` when uploading my rover's mission

Sefibrah, 2024-06-24:
I am losing hope. I've been desperately searching about this, but nothing useful is coming up.

I have a SITL Rover, and my Android device (Redmi Note 8) running my own lil' Android app, and I manage to connect them through UDP perfectly. I am subscribing to the SITL Rover's position and battery value changes, and I display them on the device screen reactively. That's running smoothly.

The problem is this: I can't upload a damn mission! I am getting timeout errors after roughly 1-3 seconds passing! Is this normal?

To make matters worse, I used QGC to upload a mission to my SITL Rover, and that worked smoothly. I downloaded the mission plan file, tried to copy it to the best of my abilities, but it's failing too...

Anyway, I hope that the information below will give you enough context to help me...

Here are my dependencies:
```
implementation 'io.mavsdk:mavsdk:2.0.1'
implementation 'io.mavsdk:mavsdk-server:2.0.0'
```

Here's my code, heavily inspired by this [method from source code](https://github.com/ArduPilot/ardupilot-mavsdk/blob/main/examples/missions_rover/missions_rover.cpp#L122):

```
public void doSomeMission() {
  List<MissionRaw.MissionItem> missionItems = new ArrayList<>();
  missionItems.add(generateMissionItem(0, 0, 16, 1, -35.3632621, 149.1652374, 585));

  // takeoff mission item, but I am Mr. rover, i don't need takeoff...
  // missionItems.add(generateMissionItem(1, 3, 22, 0, 0, 0, 0));

  missionItems.add(generateMissionItem(1, 3, 16, 0, -35.36279934, 149.16518583, 50));
  missionItems.add(generateMissionItem(2, 3, 16, 0, -35.36279844, 149.16578784, 50));
  missionItems.add(generateMissionItem(3, 3, 16, 0, -35.36321925, 149.16578136, 50));

  missionItems.add(generateMissionItem(4, 3, 20, 0, 0, 0, 0));
  Log.d(TAG, "Uploading and starting mission...");

  if(mRover != null) {
    mRover.getMissionRaw().uploadMission(missionItems)
        .doOnComplete(() -> Log.d(TAG, "Upload succeeded"))
        .doOnError(throwable -> Log.e(TAG, "Failed to upload the mission", throwable))
      .andThen(mRover.getAction().arm()
        .onErrorComplete())
      .andThen(mRover.getMissionRaw().startMission()
        .doOnComplete(() -> Log.d(TAG, "Mission started"))
        .doOnError(throwable -> Log.e(TAG, "Failed to start the mission", throwable)))
      .subscribe(() -> {
      }, throwable -> {
      });
  } else {
    Log.d(TAG, "Rover is not yet initialized!");
  }
}

public static MissionRaw.MissionItem generateMissionItem(int seq, int frame, int command, int current, double latitudeDeg, double longitudeDeg, float alt) {
  int intLatitudeDeg = (int) (latitudeDeg * 10000000);
  int intLongitudeDeg = (int) (longitudeDeg * 10000000);
  return new MissionRaw.MissionItem(seq, frame, command, current, 1, 0f, 0f, 0f, 0f, intLatitudeDeg, intLongitudeDeg, alt, 0);
}
```

Here's the error message I receive:
```
E  Failed to upload the mission
  io.mavsdk.mission_raw.MissionRaw$MissionRawException: TIMEOUT: Timeout
  at io.mavsdk.mission_raw.MissionRaw$1.onNext(MissionRaw.java:497)
  at io.mavsdk.mission_raw.MissionRaw$1.onNext(MissionRaw.java:490)
  at io.grpc.stub.ClientCalls$StreamObserverToCallListenerAdapter.onMessage(ClientCalls.java:468)
  at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1MessagesAvailable.runInternal(ClientCallImpl.java:667)
  at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1MessagesAvailable.runInContext(ClientCallImpl.java:654)
  at io.grpc.internal.ContextRunnable.run(ContextRunnable.java:37)
  at io.grpc.internal.SerializingExecutor.run(SerializingExecutor.java:133)
  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
  at java.lang.Thread.run(Thread.java:923)
```

I am writing this in hopes of help, as I am genuinely on the edge of losing my mind! Thanks to anyone who helps, and I apologise for tagging you @JonasVautherin, but you did mention that you have a lot of experience with `MissionRaw` and that you know people who use it, so I wanted to bring this to your attention first. Please forgive me.

Best Regards,
Ibrahim Sefer

julianoes, 2024-06-24:
What's running on the rover? PX4 or ArduRover?

Sefibrah, 2024-06-24:
ArduRover!

julianoes, 2024-06-24:
Ok, do you get the mavsdk_server log output? If so, can you set the env variable `MAVSDK_MISSION_TRANSFER_DEBUGGING=1`? That might give you more info.

Sefibrah, 2024-06-24:
I've noticed that it could be due to connection issues. Apparently, receiving data is not the same as sending data, right? They have their separate ports, or addresses? Does that mean that when I do this combination:

```
private void initializeServerAndDrone(@NonNull String systemAddress) {
    MavsdkEventQueue.executor().execute(() -> {
          mMavsdkServer = new MavsdkServer();
          int mavsdkServerPort = mMavsdkServer.run("udp://192.168.1.100:14550");
        
          mRover = new io.mavsdk.System("localhost", mavsdkServerPort);
          Log.d(TAG, "Rover is initialized!");
    
          // other stuff
    });
}
```

followed by

```
sim_vehicle.py -D -f gazebo-rover --model JSON --out udpin:192.168.1.100:14550 --map --console
```

I only really account for the **RECEPTION** of data, but when I try to send data (like my mission), I get `TIMEOUT: Timeout` very quickly. Could this be it? Or am I completely misinterpreting this...

Sefibrah, 2024-06-24:
> MAVSDK_MISSION_TRANSFER_DEBUGGING=1
Is this a Docker env? I am not using Docker? Just ArduRover, my laptop, and my phone.

Sefibrah, 2024-06-24:
I think I have the same issue as this guy did [here](https://github.com/mavlink/MAVSDK/issues/2211), but he says that it's fixed for him?

chehrlic, 2024-06-24:
I explictly wrote what I've  done - I've updated MAVSDK as https://github.com/mavlink/MAVSDK/commit/30d2693417fe2afdfe494ace3e0f5b8ee5331f03 fixes the issue...

Sefibrah, 2024-06-24:
But I am using `implementation 'io.mavsdk:mavsdk-server:2.0.0'` instead of `mavsdk_server` from `brew install mavsdk` like you... I guess I need to wait for the update here as well?

Sefibrah, 2024-06-24:
I posted a PR @julianoes @JonasVautherin, please check it 😭

Sefibrah, 2024-06-25:
Tried running the `mavsdk_server` locally on v2.12.2, and the mission got through! So I guess that the issue is fixed. Please accept my PR.

Sefibrah, 2024-06-26:
All that's left is just for the maintainers to push the PR to Maven, so I'll close this issue...

closed 2024-06-26 by Sefibrah