mavlink/MAVSDK-Java #171: Cannot build MAVSDK 2.1.0 on Android tablet

mfran89, 2024-07-31:
Hi ,

For some time I have been using 1.3.1 for mavsdk with out issues. I just saw that there is a 2.1.0 build and I tired to build it but get this error 

```
slf4j-api-2.0.12.jar
- java.lang.NullPointerException: Cannot invoke "String.length()" because "<parameter1>" is null
mavsdk-2.1.0.jar
- error while dexing 
```
I was wondering if anyone else had this issue ? I am still able to successfully use mavsdk 1.3.1 and mavsdk server 1.3.2. 

Thanks
Mike

JonasVautherin, 2024-08-02:
Are you including mavsdk from Maven? Or building from sources?

mfran89, 2024-08-02:
building it from maven

JonasVautherin, 2024-08-03:
Did you try updating your tooling (e.g. gradle)? I see online that this error is sometimes fixed by upgrading the tooling.

mfran89, 2024-08-06:
Currently I am using android gradle plugin version 7.4.2 and gradle 7.6.1 . I tried using both of the latest android gradle plugin 8.0.2 and gradle 8.9 but found that I couldn't even get the project to build. Havent taken a deep dive yet into what version will work best yet.

mfran89, 2024-08-14:
So I set gradle to 8.6.0 and I could only set android gradle plugin to 8.0 my build tool version 30.0.3. I used this based on your recommendation from your last merge for mavsdk 2.1.0 . My build time issue after that was 
`Property 'http://javax.xml.XMLConstants/property/accessExternalSchema' is not recognized.`
Not sure how to fix that any ideas?

mfran89, 2024-08-19:
Issue resolved , I was using flamingo android ide , and needed to upgrade it to android giraffe latest patch to then use android gradle app 8.1.2 and gradle plugin 8.2 to use mavsdk 2.1.0. Will test using later versions of gradle and android gradle plugin to see how it goes.

closed 2024-08-19 by mfran89

julianoes, 2024-08-20:
:flamingo: :giraffe: