mavlink/MAVSDK-Java #172: Having issues sending commands over TCP to drone

mfran89, 2024-08-01:
Hey!

I had used mavsdk 1.3.1 and mavsdk 1.3.2 on android  alittle while ago, primarily using it via the serial method and got it to work. I recently tried to use the tcp method of connecting to the drone and it looks like I am successfully able to connect to the drone, however I am having issues sending commands to it. 

For example I used to send 
```
List<MissionRaw.MissionItem> missionItems  = drone.getMissionRaw().downloadMission()
                    .doOnError(throwable -> Log.e(TAG, "Failed to get mission "+ throwable.getMessage())).blockingGet();
```
When I wanted to get a mission downloaded, but using TCP it blocks forever and doesnt return.

Similarly , when I want to get telemetry information like below : 
```
 _disposables.add(drone.getTelemetry().getGpsInfo().subscribe(GPS
                -> {
            String numSats = GPS.getNumSatellites().toString();
            String fixType = GPS.getFixType().toString();
            Log.d(TAG, " gps fix and type" +GPS.getNumSatellites() + "  " + GPS.getFixType());
            }, throwable -> {
            Log.e(TAG, "Error occurred while getting GPS data " + throwable.getMessage(), throwable);
        }));
```
I would usually see asynchronous log cat messages show up about the gps information. Over TCP I do not see any messages coming through. What I do see via the debug data is that MAVSDK is printing the status text like 
`MAVLink: emergency: L0: NOTICE! Auto adjusting transition alt!` but when I try to get that telemetry streamed via the subscribe method I used above I do not see that stream for via my function. 

Any thoughts and advice on this would be really helpful.

Thanks
Mike

mfran89, 2024-08-01:
A little more context ,
I am setting up the mavsdk server by setting the host address to the mavproxy --out <ip address>:<port> and then once I get a port from the mavsdk server run() I set the mavsdk system obj with the host ip address and the port returned from the server.

Thanks
Mike

mfran89, 2024-08-02:
Sorry, it was a silly issue I figured out. I was setting the ip address for System(IP_ADDRESS,PORT) to the remote IP address as opposed to the ip address of what system it was running on, in this case I set it to localhost and it worked. For some reason I found that 127.0.0.1 didn't work, but not a big deal. Hope this helps someone.

closed 2024-08-02 by mfran89

julianoes, 2024-08-04:
Thanks for the note. The connection set up will be a bit more explicit in the future: https://github.com/mavlink/MAVSDK/pull/2342