mavlink/MAVSDK-Java #173: Android app - server does not start

matthieu-db, 2024-08-05:
Hi, on the Android app when I press the 'Run MavSDK server' button, nothing happens in the app. When debugging, it seems that the `run(String systemAddress, int mavsdkServerPort)` function returns false because the call to `mavsdk_server_run` in the native cpp code (https://github.com/mavlink/MAVSDK-Java/blob/main/mavsdk_server/src/main/cpp/native-lib.cpp#L25C14-L25C31)  returns false..

Does it work for you with the code currently in the repo? What behaviour am I supposed to see here, do I need to run something else on my machine, could a firewall be the cause? From reading the code it seems the server should start on its own as it is in the example.

JonasVautherin, 2024-08-05:
How do you call it exactly? It sounds like the systemAddress may be invalid.

matthieu-db, 2024-08-05:
I just press the 'Run MavSDK server' button in the demo app (https://github.com/mavlink/MAVSDK-Java/tree/main/examples/android-client)

JonasVautherin, 2024-08-05:
It should work, yes. Do you have a drone sending MAVLink to it on UDP 14540?

> it seems that the run(String systemAddress, int mavsdkServerPort) function returns false

I am a bit confused here. Shouldn't it return an `int` (the port)? You're talking about [this](https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/app/src/main/java/io/mavsdk/androidclient/MapsActivity.java#L247), right?

matthieu-db, 2024-08-05:
Right, I meant it returns 0.

I tested it with and without a simulator running (docker, jonasvautherin/px4-gazebo-headless:1.14.3). I'm not sure how I should point the simulator to send to the app?

In any case, I think the run function should still not return 0 even without incoming MavLink on that port or am I understanding this wrong (I'm very new to the whole ecosystem)

JonasVautherin, 2024-08-05:
> I'm not sure how I should point the simulator to send to the app?

[This](https://github.com/jonasvautherin/px4-gazebo-headless?tab=readme-ov-file#run-with-a-custom-ip-for-the-second-mavlink-interface) should do it, e.g.:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 192.168.0.12
```

Where you should replace `192.168.0.12` with the IP of your Android device.

> In any case, I think the run function should still not return 0

That is correct, it should not. It seems like it [failed to start](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/mavsdk_server_api.cpp#L13-L22). Does your Android Logcat show any error? E.g. does your app have the required permissions? I haven't tried the example app in a while 😕.

matthieu-db, 2024-08-06:
Nothing really in the logs; I have tried this on a bunch of different devices with the exact same results (Android 14 phone, Android 14 emulator, Android 10 DJI controller)

This is what's in the logs (omitting the noise MapBox makes):

```
2024-08-06 08:54:29.378  2957-2957  nativeloader            io.mavsdk.androidclient              D  Load libframework-connectivity-tiramisu-jni.so using APEX ns com_android_tethering for caller /apex/com.android.tethering/javalib/framework-connectivity-t.jar: ok
2024-08-06 08:54:23.054  2957-2957  (4)[2957:k...ent]binder io.mavsdk.androidclient              I  2957:2957 ioctl 40046210 7fd30cafd4 returned -22
2024-08-06 08:54:29.399  2957-2957  ActivityThread          io.mavsdk.androidclient              I  Relaunch all activities: onCoreSettingsChange
2024-08-06 08:54:29.400  2957-2957  Compatibil...geReporter io.mavsdk.androidclient              D  Compat change id reported: 171979766; UID 10535; state: ENABLED
2024-08-06 08:54:29.400  2957-2957  Compatibil...geReporter io.mavsdk.androidclient              D  Compat change id reported: 242716250; UID 10535; state: DISABLED
2024-08-06 08:54:29.409  2957-2957  ziparchive              io.mavsdk.androidclient              W  Unable to open '/data/app/~~g7J6BOVvwuVEPS6xKeterw==/io.mavsdk.androidclient-1NsnxlPveP-bj0eA-gkDvQ==/base.dm': No such file or directory
2024-08-06 08:54:29.409  2957-2957  ziparchive              io.mavsdk.androidclient              W  Unable to open '/data/app/~~g7J6BOVvwuVEPS6xKeterw==/io.mavsdk.androidclient-1NsnxlPveP-bj0eA-gkDvQ==/base.dm': No such file or directory
2024-08-06 08:54:29.549  2957-2957  nativeloader            io.mavsdk.androidclient              D  Configuring clns-4 for other apk /data/app/~~g7J6BOVvwuVEPS6xKeterw==/io.mavsdk.androidclient-1NsnxlPveP-bj0eA-gkDvQ==/base.apk. target_sdk_version=33, uses_libraries=, library_path=/data/app/~~g7J6BOVvwuVEPS6xKeterw==/io.mavsdk.androidclient-1NsnxlPveP-bj0eA-gkDvQ==/lib/arm64:/data/app/~~g7J6BOVvwuVEPS6xKeterw==/io.mavsdk.androidclient-1NsnxlPveP-bj0eA-gkDvQ==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/io.mavsdk.androidclient
2024-08-06 08:57:20.148  2957-3207  Mavsdk                  io.mavsdk.androidclient              I  MAVSDK version: v2.12.2
2024-08-06 08:57:20.148  2957-3207  MAVSDK-Server           io.mavsdk.androidclient              D  Running mavsdk_server with connection url: udp://:14540
2024-08-06 08:57:20.148  2957-3207  Mavsdk                  io.mavsdk.androidclient              I  Waiting to discover system on udp://:14540...
```

JonasVautherin, 2024-08-06:
> Waiting to discover system on udp://:14540...

So it is running, but hasn't discovered a MAVLink system. Are you sure that your simulator is properly sending MAVLink packets to the app?

matthieu-db, 2024-08-06:
It doesn't seem to be actually running, first of all the text never changes to "Destroy Mavsdk server" as it should;
debugging it a bit, the executor never gets past the `      int mavsdkServerPort = mavsdkServer.run();
` line:

```java
  private void runMavsdkServer() {
    Log.e("TAG123" ,"button clicked");
    MavsdkEventQueue.executor().execute(() -> {
      Log.e("TAG123" ,"Begin of executor");
      int mavsdkServerPort = mavsdkServer.run();
      Log.e("TAG123" ,"got here");
      drone = new System(BACKEND_IP_ADDRESS, mavsdkServerPort);
```
This prints "Begin of executor" in the logs; but never "got here"

matthieu-db, 2024-08-06:
Okay now it works, I guess I _was_ holding it wrong, thanks for your help!

closed 2024-08-06 by matthieu-db