mavlink/MAVSDK-Java #174: connected with a real drone problem

wangpuxia, 2024-08-06:
hi

I set up a mavsdk environment on the server and tested it using jmavsim. Everything work fine
But when I connected the real drone to Mavsdk, Some functions can be used normally, such as

System drone = new System ( "123.201.28.156" , 50041 ) ;
drone.getTelemetry().getFlightMode().subscribe(flightMode -> logger.info("Flight mode: " + flightMode));

It can return the flightmode data correctly, but most functions such as

drone.getAction ( ).arm ( ) or
drone.getTelemetry ( ).getBattery ( )

do not provide any feedback. The server will output content similar to the following:

[09:49:05|Info ] MAVSDK version: v2.11.0 (mavsdk_impl.cpp:26)
[09:49:05|Info ] Waiting to discover system on udp://:14552... (connection_initiator.h:20)
[09:50:20|Info ] New system on: 39.144.112.2:48334 (with system ID: 1) (udp_connection.cpp:199)
[09:50:20|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:720)
[09:50:20|Debug] Component Autopilot (1) added. (system_impl.cpp:366)
[09:50:20|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[09:50:20|Debug] Discovered 1 component(s) (system_impl.cpp:509)
[09:50:20|Info ] System discovered (connection_initiator.h:62)
[09:50:20|Info ] Server started (grpc_server.cpp:169)
[09:50:20|Info ] Server set to listen on 0.0.0.0:50042 (grpc_server.cpp:170)
[09:50:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:50:22|Debug] MAVLink: info: u-blox 1 HW: 00080000 SW: EXT CORE 3.01 (107900) (system_impl.cpp:243)
[09:50:51|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:51|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:51:17|Warn ] sending again after 0.502024 s, retries to do: 3  (512). (mavlink_command_sender.cpp:320)
[09:51:17|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:51:18|Warn ] sending again, retries to do: 5 (mavlink_parameter_client.cpp:1057)
[09:51:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:51:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:51:51|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:51:51|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:52:13|Warn ] sending again after 0.501734 s, retries to do: 3  (512). (mavlink_command_sender.cpp:320)
[09:52:13|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:52:21|Warn ] sending again after 0.51101 s, retries to do: 3  (512). (mavlink_command_sender.cpp:320)
[09:52:21|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:52:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:52:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:52:50|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:52:50|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:53:20|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:53:20|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:53:50|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:53:50|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:54:20|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:54:20|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)

May I ask where the problem occurred? and how to fix it.

wangpuxia, 2024-08-06:
[image]

JonasVautherin, 2024-08-06:
It seems like the autopilot is refusing to arm:

```
 [09:50:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
```

wangpuxia, 2024-08-09:
> It seems like the autopilot is refusing to arm:
> 
> ```
>  [09:50:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
> [09:50:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
> ```

yes, thank you ,what should i do next?

wangpuxia, 2024-08-09:
> MAVSDK

the code is simple like this:

System drone = new System ( "101.201.28.156" , 50042 ) ;
		  drone.getAction ( ).arm ( )
		  .doOnComplete ( ( ) -> logger.debug ( "Arming..." ) )
		  .doOnError ( throwable -> {
				logger.error ( "Failed to arm: " + ( ( Action.ActionException ) throwable ).getCode ( ) );
			} ).subscribe ( latch::countDown , throwable -> latch.countDown ( ) ) ;

log message is : Failed to arm: UNKNOWN

MAVSDK console is:
[02:41:10|Warn ] sending again after 0.506188 s, retries to do: 3  (511). (mavlink_command_sender.cpp:320)
[02:41:23|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[02:41:23|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[02:41:40|Warn ] sending again after 0.505487 s, retries to do: 3  (400). (mavlink_command_sender.cpp:320)
[02:41:41|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[02:41:41|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[02:41:53|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)

JonasVautherin, 2024-08-09:
Are you running PX4 or Ardupilot?

julianoes, 2024-08-10:
```
[02:41:23|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[02:41:23|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:24
```

That's why it doesn't arm.

wangpuxia, 2024-08-10:
> Ardupilot

pixhawk 2.4.8

wangpuxia, 2024-08-10:
> ```
> [02:41:23|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
> [02:41:23|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:24
> ```
> 
> That's why it doesn't arm.

thanks, but what i do with the drone to fix " Throttle below Failsafe " problem ?

JonasVautherin, 2024-08-10:
> Ardupilot

Maybe that's why it doesn't send battery updates? @julianoes Doesn't Ardupilot require to set the message rate?

julianoes, 2024-08-10:
> Doesn't Ardupilot require to set the message rate?

I think ArduPilot requires it, yes.

wangpuxia, 2024-08-11:
> > Ardupilot
> 
> Maybe that's why it doesn't send battery updates? @julianoes Doesn't Ardupilot require to set the message rate?

i try to find out how to set message rate

wangpuxia, 2024-08-11:
but the drone connect with GPC work fine, include check battery status and takeoff ...whats the different?

JonasVautherin, 2024-08-11:
> i try to find out how to set message rate

https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/telemetry/telemetry.proto#L100

> but the drone connect with QGC work fine, include check battery status and takeoff ...whats the different?

I guess QGC sets the rate for Ardupilot?

julianoes, 2024-08-11:
> I guess QGC sets the rate for Ardupilot?

Yes it does.