mavlink/MAVSDK-Java #175: How to set Ardupilot fixed wing drone to loiter in the middle of a mission

mfran89, 2024-08-14:
Hi team,

I am currently using a Audrupilot system with mavsdk 1.3.1 and server 1.3.2. I wanted to know what the best approach was to start a mission (setting mode to "AUTO") followed by pausing the mission and then resuming while the drone is in the air.

From what I have gathered from the documentation by using the system object I can something like the following to start the mission
`
drone.getMissionRaw().startMission();
`

Then if I want to pause or loiter during the mission I saw these functions:
```
drone.getAction().hold
drone.getMission().pauseMission()
```
Which is is better? From what I understand the hold() might be geared more towards the PX4 firmware as opposed to the Ardupilot firmware. 

Finally when I want to resume the mission is it best to just use the the startMission()  function again, or is there a resume a better way to resume the mission ?

Appreciate any help I can get.

JonasVautherin, 2024-08-14:
The truth is in the C++ implementation 😊.

[`hold`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L543) sets the mode to `FlightMode::Hold`, and [`pause_mission`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L913)... does the same thing 😁. I would say that it depends on what makes your code more readable (I guess `pauseMission()` makes more sense in the context of a mission.

[`start_mission`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L896) sets the mode to `FlightMode::Mission`, so it should just resume the mission, I would think.

mfran89, 2024-08-15:
Thanks sounds good to me, for now I will use pauseMission and startMission it looks good to me via MissionPlanner .

closed 2024-08-15 by mfran89