mavlink/MAVSDK-Java #176: Data format to use RTK Plugin to forward RTCM messages

lukask0ch, 2024-08-16:
Hi, I'm trying to use MAVSDK-Java to send RTCM messages coming from an NTRIP Provider to a PX4 Drone with a F9P uBlox GPS Receiver. I get this data as a byte array and need to convert it to a string to send it with the sendRtcmData() function. 
The data is being sent and I can see it arriving in QGroundControl. However, the GPS status is still 3d lock and not going to RTK. I assume the data conversion to string is wrong. I also tried Base64 encoding, as posted here https://github.com/mavlink/MAVSDK/issues/2352. But from what I can see, the MAVSDK version used in the Java wrapper is not using Base64 encoding yet, as it has been implemented only recently https://github.com/mavlink/MAVSDK/pull/2332.
What encoding type is needed to successfully send this data?

`MAV_RTK = new Rtk("127.0.0.1",14551);`
`MAV_RTK.initialize();`
`byte[] data;  // coming from ntrip`
`Rtk.RtcmData rtkData = new Rtk.RtcmData(new String(data, StandardCharsets.UTF_8));`
`MAV_RTK.sendRtcmData(rtkData).subscribe();`

Using MAVSDK-Java 2.1.0 and PX4 v1.15.0Beta

julianoes, 2024-08-18:
You need to stick with the latest MAVSDK release if that worked. Once you update to a newer mavsdk_server (v3) which exposes the field as base64, you will have to update the Java wrapper/example. That's just how breaking releases work in my mind.

lukask0ch, 2024-08-19:
Thanks for your answer. The current Mavsdk-Java version 2.1.0 still wants a string. And I don't know how to convert it correctly. The ublox F9P does not understand the message, as shown in u-center. There was the same discussion with this on the python wrapper before (https://github.com/mavlink/MAVSDK-Python/issues/491), but no clue how to do it on java, since binary data just can not be simply stored in a string.

[image]

julianoes, 2024-08-20:
Right, so it just doesn't work until v3 is out and fixes it. Sorry about that. 

If you're blocked by this, consider [throwing something in the hat](https://github.com/sponsors/julianoes).