mavlink/MAVSDK-Java #177: Having issues with uploading mission using mavsdk 2.1

mfran89, 2024-08-20:
I am using the maven versions of mavsdk and mavsdk server 2.1 along with mavsdk event queue to upload a mission like this 
```
MavsdkEventQueue.executor().execute(() -> {
try {
                Completable res = drone.getMissionRaw().clearMission()
                        .doOnComplete(() -> Log.d(TAG, "clearMission"))
                        .doOnError(throwable -> Log.e(TAG, "Failed to clear Mission due to " + throwable.getMessage()))
                        .andThen(_blitzdrone.getMissionRaw().uploadMission(missionItems)
                                .doOnComplete(() -> Log.d(TAG, "Upload Mission succeeded"))
                                .doOnError(throwable -> Log.e(TAG, "Failed to upload the mission " + throwable.getMessage())));
                _disposables.add(res
                        .subscribe(
                                () -> {
                                    Log.d(TAG, "Mission clear and upload process completed successfully");
                                },
                                throwable -> {
                                    Log.e(TAG,"Mission clear and upload process failed: " + throwable.getMessage());
                                }
                        ));
            } catch (Throwable err) {
                Log.e(TAG, "Mission error " + err.getMessage());
            }
 });
```
The results I am getting are similar to what issue #166 implied. Which is strange since I thought the fix was implemented into the maven version that I referenced.
`Failed to clear Mission due to TIMEOUT: Timeout`
Some times the mission gets uploaded , a lot of times it does not. FWIW I am also seeing this when I try to execute actions like start mission, and pause mission . Finally I was seeing this issue on mavsdk 1.3.1 but not as much.

Am I doing something wrong in terms of writing this function?

Thanks
Mike

julianoes, 2024-08-20:
Does the communication work otherwise? That's the first step.

mfran89, 2024-08-21:
Yup the rest work I am able to connect to it and get telemetry fine.

JonasVautherin, 2024-08-21:
Did you try uploading the mission without clearing it? What autopilot is running on the other side?

Given that you receive a timeout, it could be that the autopilot just doesn't support whatever MAVSDK sends to "clear" the mission (I don't know exactly which message does that).

mfran89, 2024-08-22:
It's a generic pixhawk running Ardupilot 4.3.1. 
Some times it timeout some times I see this 
`io.mavsdk.mission_raw.MissionRaw$MissionRawException: PROTOCOL_ERROR:`
From what I read online I thought it was best to clear mission before uploading for safety purposes.

Other things I am trying is to 
1. Increase mavlink core timeout (jury is still out)
2. using blockingAwait instead of subscribing to the result , but still wrapping it in mavsdkeventqueue (has the best results but still has some hiccups ) 

So still working through some solutions right now.

JonasVautherin, 2024-08-22:
> has the best results but still has some hiccups

Do you imply here that it sometimes works? If it does, then it would hint that the system is too slow (and maybe some timeout need to be modified or, arguably, the autopilot should be faster with its ACK (if it is e.g. a command)). But if it always fails, then it could mean that the autopilot does not support the MAVLink "clearing".

Sefibrah, 2024-09-15:
@mfran89 may I ask you out of curiosity how did you manage to initialise the `drone` object? I am having trouble to do so in my own issue #178, in my android client, but it seems that you have it all figured out?

closed 2025-08-21 by JonasVautherin