mavlink/MAVSDK-Java #178: Code stops at `mavsdkServer.run(systemAddress);`

Sefibrah, 2024-09-15:
This issue may look like #173 on the surface, and this [comment](https://github.com/mavlink/MAVSDK-Java/issues/173#issuecomment-2270888828) may have the same issue like me, but I have a key difference:

My code:
```
// ...
MavsdkEventQueue
        .executor()
        .execute(
            () -> {
            Log.i(TAG, "anything? 1");
            mMavsdkServer = new MavsdkServer();
            Log.i(TAG, "anything? 2");
            int mavsdkServerPort = mMavsdkServer.run("udp://" + host + ":" + port);
            Log.i(TAG, "anything? 3");
            mRover = new io.mavsdk.System(MAVSDK_SERVER_IP, mavsdkServerPort);
            Log.i(TAG, "anything? 4");
            // ...
```

My LogCat:
```
2024-09-15 09:27:13.367 18365-18454 LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 1
2024-09-15 09:27:13.540 18365-18454 LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 2
2024-09-15 09:27:13.541 18365-18454 Mavsdk                  org...escript.karaymobilecontroller  I  MAVSDK version: v2.12.2
2024-09-15 09:27:13.541 18365-18454 MAVSDK-Server           org...escript.karaymobilecontroller  D  Running mavsdk_server with connection url: udp://192.168.1.103:14550
2024-09-15 09:27:13.541 18365-18454 Mavsdk                  org...escript.karaymobilecontroller  I  Waiting to discover system on udp://192.168.1.103:14550...
2024-09-15 09:27:14.219 18365-18458 Mavsdk                  org...escript.karaymobilecontroller  D  New system ID: 42 Comp ID: 190
2024-09-15 09:27:14.219 18365-18458 Mavsdk                  org...escript.karaymobilecontroller  D  Component Ground station (190) added.
2024-09-15 09:27:14.219 18365-18458 Mavsdk                  org...escript.karaymobilecontroller  D  Discovered 1 component(s)
```

As you can see, it did discover my component! Despite that, it refuses to continue? The simulation still works, and it got started by doing the following:

`sim_vehicle.py --model JSON --map --console -f gazebo-rover -v Rover -D --out udp:192.168.1.103:14550`

Yet, it fails to proceed... Why?

Sefibrah, 2024-09-16:
I think I made a mistake on my end, but correcting it wasn't enough.

In the previous examples, `192.168.1.103` was the IP Address of my Herelink, and it obviously could never host the SITL. So I corrected it to my laptop's address which is `192.168.1.100` on which the Rover's SITL is hosted.

I am running my SITL w/: `sim_vehicle.py --map --console --mavproxy-args="--master udp:192.168.1.100:14550" -D -f rover`

And these are the LogCat messages I receive:
```
2024-09-16 09:28:22.226 28435-28512 LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 1
2024-09-16 09:28:22.290 28435-28512 LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 2
2024-09-16 09:28:22.290 28435-28512 Mavsdk                  org...escript.karaymobilecontroller  I  MAVSDK version: v2.12.2
2024-09-16 09:28:22.290 28435-28512 MAVSDK-Server           org...escript.karaymobilecontroller  D  Running mavsdk_server with connection url: udp://192.168.1.100:14550
2024-09-16 09:28:22.290 28435-28512 Mavsdk                  org...escript.karaymobilecontroller  I  Waiting to discover system on udp://192.168.1.100:14550...
2024-09-16 09:28:22.423 28435-28517 Mavsdk                  org...escript.karaymobilecontroller  D  New system ID: 255 Comp ID: 230
2024-09-16 09:28:22.424 28435-28517 Mavsdk                  org...escript.karaymobilecontroller  D  Component Unsupported component (230) added.
2024-09-16 09:28:22.424 28435-28517 Mavsdk                  org...escript.karaymobilecontroller  D  Discovered 1 component(s)
2024-09-16 09:28:24.474 28435-28519 ProfileInstaller        org...escript.karaymobilecontroller  D  Installing profile for org.nativescript.karaymobilecontroller
2024-09-16 09:29:08.718 28435-28515 Mavsdk                  org...escript.karaymobilecontroller  I  heartbeats timed out
2024-09-16 09:29:11.458 28435-28517 Mavsdk                  org...escript.karaymobilecontroller  D  Discovered 1 component(s)
```

Seems that the `System ID` and `Comp ID` changed, but still not good enough, as it fails to get to the `Log.i(TAG, "anything? 3");` & `Log.i(TAG, "anything? 4");` logs...

What could it possibly be?

JonasVautherin, 2024-09-16:
> New system ID: 255 Comp ID: 230

Are those the sysid/compid sent by the rover? MAVSDK expects the autopilot to be 1/1, I would say

Sefibrah, 2024-09-16:
> Are those the sysid/compid sent by the rover? MAVSDK expects the autopilot to be 1/1, I would say

WDYM 1/1?

julianoes, 2024-09-17:
@JonasVautherin could it be related to the race condition between components in https://github.com/mavlink/MAVSDK/pull/2388?

julianoes, 2024-09-17:
> @Sefibrah Also do you know what does the Component Unsupported component (230) added. mean?

It means it sees component ID 230, whatever that is, but not component ID which would be the autopilot.

Not sure what 230 is, it's not listed in https://mavlink.io/en/messages/common.html#MAV_COMPONENT.

JonasVautherin, 2024-09-17:
Let's try, I updated and pushed mavsdk-server 2.1.1, which will be available through Maven in a couple hours: https://central.sonatype.com/artifact/io.mavsdk/mavsdk-server/versions

@Sefibrah: would you mind trying to update mavsdk-server to 2.1.1 and see if it helps?

Sefibrah, 2024-09-17:
I absolutely wouldn't mind at all! I'll let you know as soon as I check it out @JonasVautherin

Sefibrah, 2024-09-17:
@JonasVautherin Didn't go as expected unfortunately...

My LogCat:
```
2024-09-17 11:04:31.380  2049-2138  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 1
2024-09-17 11:04:31.566  2049-2138  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 2
2024-09-17 11:04:31.566  2049-2138  Mavsdk                  org...escript.karaymobilecontroller  I  MAVSDK version: v2.12.6
2024-09-17 11:04:31.567  2049-2138  MAVSDK-Server           org...escript.karaymobilecontroller  D  Running mavsdk_server with connection url: udp://192.168.1.100:14550
2024-09-17 11:04:31.567  2049-2138  Mavsdk                  org...escript.karaymobilecontroller  I  Waiting to discover system on udp://192.168.1.100:14550...
2024-09-17 11:04:32.190  2049-2142  Mavsdk                  org...escript.karaymobilecontroller  D  New system ID: 255 Comp ID: 230
2024-09-17 11:04:32.190  2049-2142  Mavsdk                  org...escript.karaymobilecontroller  D  Component Unsupported component (230) added.
2024-09-17 11:04:32.190  2049-2142  Mavsdk                  org...escript.karaymobilecontroller  D  Discovered 1 component(s)
```

You can see the MAVSDK version changed from `v2.12.2` to `v2.12.6`, but I guess it wasn't enough.

My dependencies:
```
dependencies {
  //  other deps...
  implementation 'io.mavsdk:mavsdk:2.1.0'
  implementation 'io.mavsdk:mavsdk-server:2.1.1'
}
```

JonasVautherin, 2024-09-17:
Do you have a component that advertises as a MAVLink autopilot? `mavsdk_server` waits until it discovers an autopilot.

Sefibrah, 2024-09-17:
Not sure, maybe the culprit is somewhere in `sim_vehicle.py --map --console --mavproxy-args="--master udp:192.168.1.100:14550" -D -f rover`, or in `sim_vehicle.py --map --console --out udp:192.168.1.100:14550 -D -f rover`.

I am trying to allow UDP access to foreign GCS (like my Herelink Custom app), but maybe I got it all wrong. I remember it working last time, but now I am not certain why is it refusing...

When I run my SITL, among other logs in the console I get:

```
// ...
AP: ArduPilot Ready
AP: 192.168.1.100
// ...
```

It took the `--out` / `--mavproxy-args` flags that I provided it, so it must be emitting some form of UDP data right?

Sefibrah, 2024-09-17:
I am happy, but feeling dumb at the same time. I found the resolution to the problem:

```
sim_vehicle.py --map --console --out udpin:192.168.1.100:14550 -D -f rover
```

Instead of using `udp` I needed to use `udpin`...

Sometimes this sh*t drives me nuts 😅

Thank you 2 for helping me out. Btw here are how the logs look now.

My LogCat:
```
2024-09-17 17:55:58.829  3106-3168  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 1
2024-09-17 17:55:58.974  3106-3168  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 2
2024-09-17 17:55:58.974  3106-3168  Mavsdk                  org...escript.karaymobilecontroller  I  MAVSDK version: v2.12.6
2024-09-17 17:55:58.975  3106-3168  MAVSDK-Server           org...escript.karaymobilecontroller  D  Running mavsdk_server with connection url: udp://192.168.1.100:14550
2024-09-17 17:55:58.975  3106-3168  Mavsdk                  org...escript.karaymobilecontroller  I  Waiting to discover system on udp://192.168.1.100:14550...
2024-09-17 17:55:59.022  3106-3171  Mavsdk                  org...escript.karaymobilecontroller  D  New system ID: 1 Comp ID: 1
2024-09-17 17:55:59.022  3106-3171  Mavsdk                  org...escript.karaymobilecontroller  D  Component Autopilot (1) added.
2024-09-17 17:55:59.076  3106-3168  Mavsdk                  org...escript.karaymobilecontroller  I  System discovered
2024-09-17 17:55:59.080  3106-3168  Mavsdk                  org...escript.karaymobilecontroller  I  Server started
2024-09-17 17:55:59.080  3106-3168  Mavsdk                  org...escript.karaymobilecontroller  I  Server set to listen on 0.0.0.0:41555
2024-09-17 17:55:59.081  3106-3168  MAVSDK-Server           org...escript.karaymobilecontroller  D  mavsdk_server is now running, listening on port 41555
2024-09-17 17:55:59.081  3106-3168  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 3
2024-09-17 17:55:59.083  3106-3168  LOG_Mavsdk...munication org...escript.karaymobilecontroller  I  anything? 4
```

Thanks once again... 😭

closed 2024-09-17 by Sefibrah