mavlink/MAVSDK-Java #180: MAVSDK 2.1.0 Seg Fault Crashes When Tearing Down Server

rayw-dronesense, 2024-09-19:
## Environment

- Herelink Controller (Android 7.1.2)
- Gazebo Headless Simulator (https://github.com/JonasVautherin/px4-gazebo-headless)

## Repro
1. Install example app (attached as [ReproMavsdkCrash.zip](https://github.com/user-attachments/files/17066260/ReproMavsdkCrash.zip))
2. Start the server
3. Connect the Simulator to the app (on port 14540)
4. Wait until aircraft attitude is shown in Logcat
5. Click "stop server"
6. Crash

## Stack trace

```
2024-09-19 16:36:17.585 14170-14336 MAVLinkService          com.example.repromavcrash            E  Got attitude from aircraft: -1.2388214 -1.231313 93.332565
2024-09-19 16:36:17.595 14170-14336 MAVLinkService          com.example.repromavcrash            E  Got attitude from aircraft: -1.2398055 -1.2350508 93.33456
2024-09-19 16:36:17.601 14170-14336 MAVLinkService          com.example.repromavcrash            E  Got attitude from aircraft: -1.2399287 -1.2368649 93.33481
2024-09-19 16:36:17.631   273-13156 MVE_OMX_IF              media.codec                          E  Buffer abandoned. buffer_id 6f9. (mve_omx_common_component.c:1987)
2024-09-19 16:36:17.700 14170-14208 MAVLinkService          com.example.repromavcrash            E  Server 14551 no longer active.
2024-09-19 16:36:17.700 14170-14215 libc                    com.example.repromavcrash            A  Fatal signal 11 (SIGSEGV), code 1, fault addr 0x16a in tid 14215 (DefaultDispatch)
2024-09-19 16:36:17.701   230-230   <no-tag>                debuggerd64                          W  debuggerd: handling request: pid=14170 uid=10064 gid=10064 tid=14215
2024-09-19 16:36:17.739 14347-14347 DEBUG                   pid-14347                            A  *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2024-09-19 16:36:17.739 14347-14347 DEBUG                   pid-14347                            A  Build fingerprint: 'CubePilot/song/song:7.1.2/N2G47J/RU01240522:user/test-keys'
2024-09-19 16:36:17.739 14347-14347 DEBUG                   pid-14347                            A  Revision: '0'
2024-09-19 16:36:17.739 14347-14347 DEBUG                   pid-14347                            A  ABI: 'arm64'
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A  pid: 14170, tid: 14215, name: DefaultDispatch  >>> com.example.repromavcrash <<<
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A  signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x16a
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x0   000000000000016a  x1   00000074d905f398  x2   0000000000000000  x3   00000074d8f5d4e8
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x4   00000074d8f5d460  x5   00000075146bef4c  x6   00000074d8f5d450  x7   ffffffffffffffff
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x8   00000074d905f3d0  x9   fc83543f6d416706  x10  ffffffffc4653600  x11  0000000000000000
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x12  0000000000000001  x13  00000074d905ef88  x14  0000000000000041  x15  0000000000000000
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x16  00000074db55b220  x17  00000075146bf968  x18  000000750fbf07e0  x19  000000000000016a
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x20  00000074d905f3d0  x21  0000000000000022  x22  0000000000000000  x23  0000000000000000
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x24  00000074db2d5928  x25  00000074d905f3e8  x26  00000074db2d1258  x27  00000074db2d1398
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      x28  00000074db2d1450  x29  00000074d905f380  x30  00000074db526ee0
2024-09-19 16:36:17.740 14347-14347 DEBUG                   pid-14347                            A      sp   00000074d905f380  pc   00000075146bf968  pstate 0000000060000000
2024-09-19 16:36:17.743 14347-14347 DEBUG                   pid-14347                            A  
                                                                                                    backtrace:
2024-09-19 16:36:17.743 14347-14347 DEBUG                   pid-14347                            A      #00 pc 0000000000068968  /system/lib64/libc.so (pthread_mutex_lock)
2024-09-19 16:36:17.743 14347-14347 DEBUG                   pid-14347                            A      #01 pc 00000000000c5edc  /data/app/com.example.repromavcrash-1/base.apk (offset 0x8000)
```

julianoes, 2024-09-21:
Hm, any chance you can get a backtrace with more info? It looks like it happens in some lock but I'm not sure which one and when.

rayw-dronesense, 2024-09-23:
Sure, here's the Tombstone for it

[tombstone_09.zip](https://github.com/user-attachments/files/17100464/tombstone_09.zip)

rayw-dronesense, 2024-09-23:
Slightly different tombstone for it

[tombstone_07.zip](https://github.com/user-attachments/files/17100481/tombstone_07.zip)

rayw-dronesense, 2024-10-03:
So I nailed down the commit that introduced this particular crash, went through the different versions on the v2.12 branch on MAVSDK and saw that this: https://github.com/mavlink/MAVSDK/commit/71126ac267a10d1c5a7fcb7a3c963688b2b32277

Reverting this change while on the 2.12.6 tag made this particular crash go away.

But I do see the intent of the change was to improve connecting to systems, so we probably don't want to remove this but instead fix this somehow...

julianoes, 2024-10-04:
Great find, sorry for being slow here.

Given you seem to be able to build the server and try out different versions, any chance you can build debug mode and get more output where it crashes?

julianoes, 2024-10-04:
Or if you have hints how I can reproduce it with the [example apps](https://github.com/mavlink/MAVSDK-Java/tree/main/examples) that would work too.

rayw-dronesense, 2024-10-04:
Hi @julianoes -  I attached a custom example app (https://github.com/user-attachments/files/17066260/ReproMavsdkCrash.zip) that can easily repro the issue.

It can be done by,

1. Start example app
2. Click "start server"
3. Connect the Gazeo-Headless-PX4 sim to the app
4. wait for attitude to start showing up
5. Hit the "stop server" button
6. Reproduced crash

julianoes, 2024-10-05:
Thanks for that. I can reproduce it, but I'm not sure what to make of it yet.

It looks like it crashes here:
https://github.com/mavlink/MAVSDK/blob/27c5c30749076d205f8f5e46622171051b255386/src/mavsdk/core/log.h#L70-L83

julianoes, 2024-10-06:
Ok, so now I realized that you're starting 3 servers, one on each UDP port. Are you actually connecting 3 vehicles? Or is this just to connect three ports? If it is to do 3 ports but one vehicle, you might want to add some [mavlink-router](https://github.com/mavlink-router/mavlink-router) or [MAVSDK based forwarder](https://github.com/mavlink/MAVSDK/tree/main/examples/mavlink_forwarding) in-between.

In any case, the crash happens for the one where nothing is connected! I'll have a closer look into this.

julianoes, 2024-10-06:
I've finally managed to copy in my debug build mavsdk_server.so and added some printfs.

It turns out the `_should_exit` flag here is not set to `true`, for whatever mystery reason:

https://github.com/mavlink/MAVSDK/blob/v2.12/src/mavsdk_server/src/connection_initiator.h#L35

julianoes, 2024-10-06:
After plenty of head scratching I figured out what was going on, here is the fix: https://github.com/mavlink/MAVSDK/pull/2417.

julianoes, 2024-10-07:
Fix coming with https://github.com/mavlink/MAVSDK/releases/tag/v2.12.9.

@JonasVautherin do you mind making a Java release once the artifacts are ready? Thanks!

JonasVautherin, 2024-10-07:
I guess I should wait for 2.12.10 with https://github.com/mavlink/MAVSDK/pull/2421?

julianoes, 2024-10-07:
@JonasVautherin no https://github.com/mavlink/MAVSDK/pull/2421 is only on main, not v2.12.

JonasVautherin, 2024-10-08:
That's done in mavsdk_server:2.1.3 :+1:

daniel-sales, 2024-10-08:
I see some crashes with "ReproMavsdkCrash" app even after bumping mavsdk-server to 2.1.3.
Crashes happen after some consecutive server start/stop calls, or if app is relaunched after a successful drone connection without stopping the server first.

julianoes, 2024-10-09:
@daniel-sales ok I will have to try to reproduce that again.

> Ok, so now I realized that you're starting 3 servers, one on each UDP port. Are you actually connecting 3 vehicles? Or is this just to connect three ports? If it is to do 3 ports but one vehicle, [...]

I'm still curious on this @rayw-dronesense and @daniel-sales.

daniel-sales, 2024-10-09:
@julianoes On my end I'm starting only one server on port (14551).

Regarding the connection issue, it is reproductible with the @rayw-dronesense's example app if you:

1. launch the app (I left only 14551 on `ports` list at `MAVLinkService.kt`)
2. start server
3. wait for drone connection and attitude logs 
4. close the app without hitting "stop server"
5. relaunch the app

Here's the stack trace:
```
2024-10-08 09:22:36.406  5635-5718  Mavsdk                  com.example.repromavcrash            I  MAVSDK version: v2.12.9
2024-10-08 09:22:36.414  5635-5718  MAVSDK-Server           com.example.repromavcrash            D  Running mavsdk_server with connection url: udp://:14551
2024-10-08 09:22:36.418  5635-5991  Mavsdk                  com.example.repromavcrash            I  New system on: 127.0.0.1:47005 (with system ID: 1)
2024-10-08 09:22:36.418  5635-5991  Mavsdk                  com.example.repromavcrash            D  New system ID: 1 Comp ID: 1
2024-10-08 09:22:36.425  5635-5991  Mavsdk                  com.example.repromavcrash            D  Component Autopilot (1) added.
2024-10-08 09:22:36.444  5635-5991  Mavsdk                  com.example.repromavcrash            D  New system ID: 42 Comp ID: 190
2024-10-08 09:22:36.449  5635-5991  Mavsdk                  com.example.repromavcrash            D  Component Ground station (190) added.
2024-10-08 09:22:36.449  5635-5991  Mavsdk                  com.example.repromavcrash            D  Discovered 1 component(s)
2024-10-08 09:22:36.510  5635-5991  Mavsdk                  com.example.repromavcrash            D  Component Camera 1 (100) added.
2024-10-08 09:22:36.516  5635-5718  Mavsdk                  com.example.repromavcrash            I  System discovered
2024-10-08 09:22:36.529  5635-5718  Mavsdk                  com.example.repromavcrash            I  Server started
2024-10-08 09:22:36.529  5635-5718  Mavsdk                  com.example.repromavcrash            I  Server set to listen on 0.0.0.0:42723
2024-10-08 09:22:36.529  5635-5718  MAVSDK-Server           com.example.repromavcrash            D  mavsdk_server is now running, listening on port 42723
2024-10-08 09:22:36.530  5635-5718  MAVLinkService          com.example.repromavcrash            E  Got drone connection io.mavsdk.System@e7d61f5 on port 42723
2024-10-08 09:22:36.552  5635-5748  MAVLinkService          com.example.repromavcrash            E  Connection state changed: false
2024-10-08 09:22:36.554  5635-5748  MAVLinkService          com.example.repromavcrash            E  Connection state changed: true
2024-10-08 09:22:36.557  5635-5748  MAVLinkService          com.example.repromavcrash            E  Connection state changed: false
2024-10-08 09:22:36.558  5635-5748  MAVLinkService          com.example.repromavcrash            E  Connection state changed: true
2024-10-08 09:22:36.580  5635-6011  libc                    com.example.repromavcrash            A  Fatal signal 6 (SIGABRT), code -6 in tid 6011 (DefaultDispatch)
```

julianoes, 2024-10-10:
Ok, so this is a SIGABORT and happens when something is _connected_, not disconnected like before.

julianoes, 2024-10-10:
I can't reproduce this just yet. When you say "close the app", how do you do that? I'm using the emulator and hit stop, then I can just relaunch.

My guess is that you run into a bind issue with port already bound by the previous run.

daniel-sales, 2024-10-10:
> I can't reproduce this just yet. When you say "close the app", how do you do that?

Hitting AndroidStudio "stop" button or using Android device UI to close the app. App crashes on next "start server" attempt.

> My guess is that you run into a bind issue with port already bound by the previous run.

That makes sense. The crashes go away when I revert [PR#2388](https://github.com/mavlink/MAVSDK/pull/2388), so maybe the new autopilot discovery method is interfering on the automatic port release by system when app is destroyed?

julianoes, 2024-10-10:
I honestly don't see how that would be related. And I could not reproduce it that way I'm afraid.

daniel-sales, 2024-10-11:
I just made more tests here and indeed, issue is not reproductible with px4-gazebo-headless simulator.
However, it is easily reproductible on my real drone setup. So here is the detailed info for more reliable repro steps:

- Herelink Controller (Android 7.1.2)
- Devkit drone (CubeOrange+ with Ardupilot 4.5.3, Herelink v1.1 and Siyi A8 mini gimbal)
- ReproMavsdkCrash app (opening only port 14551), mavsdk 2.1.0 and mavsdk-server 2.1.3

Crash happening on "start server" command after a few start/stop loops, or on next "start server" after closing the app without manual server stop.

**Stack trace**
``` 
2024-10-11 10:52:57.492  8080-8267  Mavsdk                  com.example.repromavcrash            I  MAVSDK version: v2.12.9
2024-10-11 10:52:57.493  8080-8267  MAVSDK-Server           com.example.repromavcrash            D  Running mavsdk_server with connection url: udp://:14551
2024-10-11 10:52:57.502  8080-8272  Mavsdk                  com.example.repromavcrash            I  New system on: 127.0.0.1:43748 (with system ID: 1)
2024-10-11 10:52:57.502  8080-8272  Mavsdk                  com.example.repromavcrash            D  New system ID: 1 Comp ID: 1
2024-10-11 10:52:57.503  8080-8272  Mavsdk                  com.example.repromavcrash            D  Component Autopilot (1) added.
2024-10-11 10:52:57.529  8080-8272  Mavsdk                  com.example.repromavcrash            D  New system ID: 42 Comp ID: 190
2024-10-11 10:52:57.530  8080-8272  Mavsdk                  com.example.repromavcrash            D  Component Ground station (190) added.
2024-10-11 10:52:57.530  8080-8272  Mavsdk                  com.example.repromavcrash            D  Discovered 1 component(s)
2024-10-11 10:52:57.595  8080-8267  Mavsdk                  com.example.repromavcrash            I  System discovered
2024-10-11 10:52:57.611  8080-8272  Mavsdk                  com.example.repromavcrash            D  Component Unsupported component (68) added.
2024-10-11 10:52:57.614  8080-8272  Mavsdk                  com.example.repromavcrash            D  Component Camera 1 (100) added.
2024-10-11 10:52:57.628  8080-8267  Mavsdk                  com.example.repromavcrash            I  Server started
2024-10-11 10:52:57.628  8080-8267  Mavsdk                  com.example.repromavcrash            I  Server set to listen on 0.0.0.0:34613
2024-10-11 10:52:57.628  8080-8267  MAVSDK-Server           com.example.repromavcrash            D  mavsdk_server is now running, listening on port 34613
2024-10-11 10:52:57.631  8080-8267  MAVLinkService          com.example.repromavcrash            E  Got drone connection io.mavsdk.System@5322373 on port 34613
2024-10-11 10:52:57.641  8080-8267  System.err              com.example.repromavcrash            W  SLF4J(W): No SLF4J providers were found.
2024-10-11 10:52:57.641  8080-8267  System.err              com.example.repromavcrash            W  SLF4J(W): Defaulting to no-operation (NOP) logger implementation
2024-10-11 10:52:57.641  8080-8267  System.err              com.example.repromavcrash            W  SLF4J(W): See https://www.slf4j.org/codes.html#noProviders for further details.
2024-10-11 10:52:57.705  8080-8292  art                     com.example.repromavcrash            W  Before Android 4.1, method double java.util.concurrent.ThreadLocalRandom.internalNextDouble(double, double) would have incorrectly overridden the package-private method in java.util.Random
2024-10-11 10:52:57.705  8080-8292  art                     com.example.repromavcrash            W  Before Android 4.1, method int java.util.concurrent.ThreadLocalRandom.internalNextInt(int, int) would have incorrectly overridden the package-private method in java.util.Random
2024-10-11 10:52:57.705  8080-8292  art                     com.example.repromavcrash            W  Before Android 4.1, method long java.util.concurrent.ThreadLocalRandom.internalNextLong(long, long) would have incorrectly overridden the package-private method in java.util.Random
2024-10-11 10:52:57.785  8080-8296  MAVLinkService          com.example.repromavcrash            E  Connection state changed: false
2024-10-11 10:52:57.787  8080-8296  MAVLinkService          com.example.repromavcrash            E  Connection state changed: true
2024-10-11 10:52:57.795  8080-8296  MAVLinkService          com.example.repromavcrash            E  Connection state changed: false
2024-10-11 10:52:57.796   296-321   BoardControl            pid-296                              E  Unable to create sockfd: Connection refused
2024-10-11 10:52:57.799  8080-8296  MAVLinkService          com.example.repromavcrash            E  Connection state changed: true
2024-10-11 10:52:57.826  8080-8299  libc                    com.example.repromavcrash            A  Fatal signal 6 (SIGABRT), code -6 in tid 8299 (DefaultDispatch)
2024-10-11 10:52:57.827   232-232   <no-tag>                pid-232                              W  debuggerd: handling request: pid=8080 uid=10064 gid=10064 tid=8299
2024-10-11 10:52:57.913  8302-8302  DEBUG                   pid-8302                             A  *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2024-10-11 10:52:57.914  8302-8302  DEBUG                   pid-8302                             A  Build fingerprint: 'CubePilot/song/song:7.1.2/N2G47J/RU01240301:user/test-keys'
2024-10-11 10:52:57.914  8302-8302  DEBUG                   pid-8302                             A  Revision: '0'
[... cut ...]
2024-10-11 10:52:57.930  8302-8302  DEBUG                   pid-8302                             A      #02 pc 0000000000023d68  /system/lib64/libc.so (raise+24)
2024-10-11 10:52:57.930  8302-8302  DEBUG                   pid-8302                             A      #03 pc 000000000001c7ec  /system/lib64/libc.so (abort+52)
2024-10-11 10:52:57.930  8302-8302  DEBUG                   pid-8302                             A      #04 pc 0000000000003e0c  /data/app/com.example.repromavcrash-1/base.apk (offset 0x147d000)
---------------------------- PROCESS ENDED (8080) for package com.example.repromavcrash ----------------------------
```

julianoes, 2024-10-11:
Ok, any chance you can debug this yourself by building the .so file in debug mode and copying it in as these instructions:
https://github.com/mavlink/MAVSDK-Java?tab=readme-ov-file#debugging-without-pushing-to-maven

That's essentially what I did to debug this. It should hopefully tell us which abort call is happening.

AlexShafir, 2024-11-16:
I also have crash after I stop mavsdk server on Android. If I later call destroy, than it works ok, but next connect takes several seconds.

I connect to latest ArduCopter SITL built with Mission Planner.
I connect not directly, but I have server port to imitate ip connection and client port to talk with SITL (imitates USB connection).

Libs:
io.mavsdk:mavsdk:2.1.0
io.mavsdk:mavsdk-server:2.1.3

Phone: Xioami Redmi Note 8T
OS: Android 9

```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2024-11-16 15:19:24.502 10601-10601 DEBUG                   pid-10601                            A  Build fingerprint: 'xiaomi/willow/willow:9/PKQ1.190616.001/V11.0.5.0.PCXMIXM:user/release-keys'
2024-11-16 15:19:24.502 10601-10601 DEBUG                   pid-10601                            A  Revision: '0'
2024-11-16 15:19:24.502 10601-10601 DEBUG                   pid-10601                            A  ABI: 'arm64'
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A  pid: 10375, tid: 10512, name: DefaultDispatch  >>> com.dev.drone <<<
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A  signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7c661250e0
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x0  0000000000000000  x1  0000000000000000  x2  0000000000000000  x3  0000000000000004
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x4  0000000000000071  x5  00000000ffffffff  x6  0000007d143d5000  x7  00000000080cbc58
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x8  0000000000000000  x9  0000000000000000  x10 0000000000000001  x11 0000000000000000
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x12 0000007c6e04b5f0  x13 0000000000000000  x14 0000007c7fdc0ae9  x15 0000000000000000
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x16 0000007c69b8a220  x17 0000007d13519f74  x18 0000000000000010  x19 0000007c6de68040
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x20 0000007c661250e0  x21 0000007c87b3b480  x22 0000007c87b3b498  x23 0000007c6dda4f90
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x24 0000007c6dda4fa0  x25 0000007c673ff438  x26 0000007c673ff390  x27 0000007c6bcf0560
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      x28 0000007c6bc7e000  x29 0000007c673ff350
2024-11-16 15:19:24.503 10601-10601 DEBUG                   pid-10601                            A      sp  0000007c673ff310  lr  0000007c6af6a108  pc  0000007c6af6a108
2024-11-16 15:19:24.597 10601-10601 DEBUG                   pid-10601                            A  
                                                                                                    backtrace:
2024-11-16 15:19:24.597 10601-10601 DEBUG                   pid-10601                            A      #00 pc 0000000001c8b108  /data/app/com.dev.drone-nAK5xUJX6A6y87cF67SonQ==/base.apk (offset 0x1c89000)
```

JonasVautherin, 2025-08-21:
It's been inactive for a while and we've been on MAVSDK 3 for months. Let's close this.

closed 2025-08-21 by JonasVautherin