mavlink/MAVSDK-Java #182: MavSDK-JAVA v2.0.0, INVALID_SEQUENCE for uploading mission with MissionRaw

Ba0Nguyen, 2024-09-27:
I am unable to upload a mission using MissionRaw. My mission have multiple MissionRaw.MissionItem, sample code below:

`
private fun createMissionPlan(): missionPlan: List<MissionRaw.MissionItem> { 

val missionRawItems = mutableListOf<MissionRaw.MissionItem>()

// Set Home Point
missionRawItems.add(createMissionItemRaw(0, 1, 16, 1, 1,0f, 0f, 0f, 0f,0f, 0f,548.6f, 0))

// Take off
missionRawItems.add(createMissionItemRaw(1, 0, 22, 0, 1,0f, 0f, 0f, 0f,0f, 0f, altitudeM.toFloat(), 0))

// add two waypoints
missionRawItems.add(createMissionItemRaw(2, 0, 16, 0, 1,0f, 0f, 0f, 0f, latitude, longitude, altitude, 0))
missionRawItems.add(createMissionItemRaw(3, 0, 16, 0, 1,0f, 0f, 0f, 0f, latitude, longitude, altitude, 0))
 
return missionRawItems  
}

 private fun createMissionItemRaw(seq: Int, frame: Int, command: Int, current: Int, autocontinue: Int,
                                  param1: Float, param2: Float, param3: Float, param4: Float,
                                  x: Float, y: Float, z: Float, missionType: Int): MissionRaw.MissionItem {

        val missionItem = MissionRaw.MissionItem(seq, frame, command, current, autocontinue,
            param1, param2, param3, param4,
            (x * 1e7).roundToInt(), (y * 1e7).roundToInt(), z, missionType)

        return missionItem
    }
`

I am able to upload the mission successfully with just only one MissionItem (e.g. home point). Whenever I try to add a mission with two MissionItems or more, I got Invalid_Sequence even thought my sequence number is correctly increment from 0.

Thanks for your help!

julianoes, 2024-09-30:
Is this against PX4 or ArduPilot?

Can you make sure that Current is 0 for the 0th item, and 1 for the first one?

Also see this thread: https://github.com/mavlink/MAVSDK/issues/2058

Ba0Nguyen, 2024-09-30:
This is against Ardupilot v4.4.4.

Ba0Nguyen, 2024-09-30:
`// set home position
        missionRawItems.add(createMissionItemRaw(0, 5, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473977420f, 85455939f,488.7872009277344f, 0))
        // set take off
        missionRawItems.add(createMissionItemRaw(0, 3, 22, 0, 1,
            0f, 0f, 0f, 0f,
            473977420f, 85455939f, 50f, 0))
        
        // set waypoints
        missionRawItems.add(createMissionItemRaw(1, 3, 16, 1, 1,
            0f, 0f, 0f, 0f,
            waypoints[5].latitude.toFloat(),  waypoints[5].longitude.toFloat(), altitudeM.toFloat(), 0))

        missionRawItems.add(createMissionItemRaw(2, 3, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473982493f, 85449242f,50f, 0))

        missionRawItems.add(createMissionItemRaw(3, 3, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473977416f,85449242f, 50f, 0))

        missionRawItems.add(createMissionItemRaw(4, 2, 20, 0, 1,
            0f, 0f, 0f, 0f,
            0f, 0f, 0f, 0))`
            
            I tried this and got "Invalid Sequence INVALID_SEQUENCE"

Ba0Nguyen, 2024-09-30:
`// set home position
        missionRawItems.add(createMissionItemRaw(0, 5, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473977420f, 85455939f,488.7872009277344f, 0))
        // set take off
        missionRawItems.add(createMissionItemRaw(0, 3, 22, 1, 1,
            0f, 0f, 0f, 0f,
            473977420f, 85455939f, 50f, 0))

        // set waypoints
        missionRawItems.add(createMissionItemRaw(1, 3, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473982266f, 85455604f, 50f, 0))

        missionRawItems.add(createMissionItemRaw(2, 3, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473982493f, 85449242f,50f, 0))

        missionRawItems.add(createMissionItemRaw(3, 3, 16, 0, 1,
            0f, 0f, 0f, 0f,
            473977416f,85449242f, 50f, 0))

        missionRawItems.add(createMissionItemRaw(4, 2, 20, 0, 1,
            0f, 0f, 0f, 0f,
            0f, 0f, 0f, 0))`

I also tried the above and got Invalid Sequence INVALID_SEQUENCE

Ba0Nguyen, 2024-09-30:
I also tried 0,1,2,3,4 sequence and got INVALID_ARGUMENT.

Ba0Nguyen, 2024-09-30:
I am logging out messages MavSDK is receiving from Ardupilot. What I notice is I only received MISSION_REQUEST_INT for 0, which explain why I was able to upload mission with 1 Mission Item successfully. I've never receive MISSION_REQUEST_INT for 1 and failed to upload mission with 2 or more Mission Items.

Ba0Nguyen, 2024-09-30:
Do you think this might potentially giving us problem?
       ` // ArduCopter 3.6 sends MISSION_REQUEST (not _INT) but actually accepts ITEM_INT in reply

        // FIXME: this will mess with the sequence number.
        mavlink_message_t request_int_message;
        mavlink_msg_mission_request_int_pack(
            request_message.sysid,
            request_message.compid,
            &request_int_message,
            request.target_system,
            request.target_component,
            request.seq,
            request.mission_type);`

julianoes, 2024-10-01:
Can you provide an example, or add a mission example to the example app? That way I can easily try to reproduce this issue.

Ba0Nguyen, 2024-10-01:
I got this as result "ERROR: INVALID_SEQUENCE: Invalid Sequence null" for the above code.

julianoes, 2024-10-01:
What does this do?

```
        var seq = missionPlan.size // size = 2
```

julianoes, 2024-10-01:
You write v2.0.0 but that version doesn't exist, so what version exactly are you using?

Ba0Nguyen, 2024-10-01:
Sorry, I meant v2.0.1

julianoes, 2024-10-01:
v2.1.0 maybe?
https://github.com/mavlink/MAVSDK-Java/releases

Ba0Nguyen, 2024-10-01:
To clarify:

MAVSDK-JAVA (I pulled from the main branch in May, so it wasn't an official release yet)
- mavsdk-server: v2.1.0 
- sdk: v2.0.1

Ba0Nguyen, 2024-10-02:
While debugging, I noticed that the unregister_all function in the MavlinkMessageHandler class (in MAVSDK C++) is called when I try to upload the mission to the drone flight controller. This did not happen with the simulator and I was able to upload the flight plan successfully.

julianoes, 2024-10-03:
Hm, so this issue still persists for SITL then?

Ba0Nguyen, 2024-10-03:
It is working when I upload mission to SITL.

julianoes, 2024-10-04:
Nice ok, so I can close this.

Ba0Nguyen, 2024-10-07:
To clarify :

It is always working with SITL but I've never get a mission to upload successfully on the actually Flight Controller with Ardupilot v4.4.4. I always got INVALID_SEQUENCE when try to upload to the actual Flight Controller.

Ba0Nguyen, 2024-10-15:
Were you able to reproduce the issue?

julianoes, 2024-10-15:
Not yet, thanks for the reminder. Easiest would be if you can make a branch of the example app that shows it.

Ba0Nguyen, 2024-11-04:
Sorry for the delay, I created a sample android app with the implementation that I have for uploading mission.

https://github.com/flybyfood/MavsdkUploadMissionSampleApp

julianoes, 2024-11-06:
@Ba0Nguyen can you tell me what commands you run to run SITL, just so I'm on the same vehicle etc., thanks!

Ba0Nguyen, 2024-11-06:
./Tools/autotest/sim_vehicle.py -v ArduCopter --console --map --out=<WIFI_IP>:<PORT>

julianoes, 2024-11-08:
Ok, thanks. I've tried that and I get:

```
2024-11-08 13:24:40.383  5562-5585  Mavsdk                  com...ple.mavsdkuploadmissionsample  D  MAVLink: info: Flight plan received
2024-11-08 13:24:40.396  5562-5628  Mavsdk-Upload-Mission   com...ple.mavsdkuploadmissionsample  D  Upload mission success.
2024-11-08 13:24:40.527  5562-5634  ProfileInstaller        com...ple.mavsdkuploadmissionsample  D  Installing profile for com.example.mavsdkuploadmissionsample
```

Ba0Nguyen, 2024-11-08:
I tried that and it has always worked for me when I run it with SITL. When I tried uploading missions to the actual Cube Orange Flight Controller, I always got INVALID_SEQUENCE.

julianoes, 2024-11-10:
Oh, I'm sorry, so I need to connect to a real controller!

julianoes, 2024-11-10:
Ok, connected with emulator via mavlink-forwarding (using MAVSDK) to CubeOrange (running ArduCopter latest master) via USB.

And I get:
```
2024-11-11 09:06:29.528  6640-6661  Mavsdk                  com...ple.mavsdkuploadmissionsample  W  timeout: retries exceeded
2024-11-11 09:06:29.537  6640-6694  Mavsdk-Upload-Mission   com...ple.mavsdkuploadmissionsample  D  Upload Error: TIMEOUT: Timeout
```

Which is not good but that's not INVALID_SEQUENCE.

julianoes, 2024-11-10:
Ok, now I could suddenly reproduce it:
```
2024-11-11 11:00:23.733  9222-9272  Mavsdk-Upload-Mission   com...ple.mavsdkuploadmissionsample  D  Upload Error: INVALID_SEQUENCE: Invalid Sequence
```

julianoes, 2024-11-11:
I think it comes down to the link. If there happens to be two links between ArduPilot and MAVSDK, and the messages get duplicated, then ArduPilot will respond with INVALID_SEQUENCE to a duplicate MISSION_REQUEST_INT. I think instead it should just ignore it.

julianoes, 2024-11-11:
I think this would be the patch that I will suggest for ArduPilot:

```
diff --git a/libraries/GCS_MAVLink/MissionItemProtocol.cpp b/libraries/GCS_MAVLink/MissionItemProtocol.cpp
index 5667578ad0..c2f92bd4e0 100644
--- a/libraries/GCS_MAVLink/MissionItemProtocol.cpp
+++ b/libraries/GCS_MAVLink/MissionItemProtocol.cpp
@@ -271,7 +271,11 @@ void MissionItemProtocol::handle_mission_item(const mavlink_message_t &msg, cons
     }
 
     // check if this is the requested waypoint
-    if (cmd.seq != request_i) {
+    if (cmd.seq < request_i) {
+        // ignore duplicate requests that we have already served earlier
+        return;
+    } else if (cmd.seq > request_i) {
+        // complain if a request has skipped an item
         send_mission_ack(msg, MAV_MISSION_INVALID_SEQUENCE);
         return;
     }
```

Any chance you could test it?

Ba0Nguyen, 2024-11-12:
I tried your Ardupilot patch and the mission uploads successfully. Do you know why there are 2 links between Ardupilot and MAVSDK? Is there only 1 link when we connect to the simulator? Is there any way we can avoid having to change Ardupilot to upload these missions?

julianoes, 2024-11-13:
I'm not sure. It shouldn't happen in general. What's your connection setup?

Ba0Nguyen, 2024-11-13:
It's a pretty simple setup. We have a Herelink air unit connected to the flight controller. The Air unit talks to a Herelink controller which is running the Android app. They're both powered and nothing else is plugged into the flight controller.

julianoes, 2024-11-13:
Ok, so I should be able to reproduce it with my Herelink then.

julianoes, 2024-11-13:
Ok, I gave it a try and could reproduce the behavior. 

It turns out it creates two connections because of the connection string `udp://localhost:14552`. If you use `udp://127.0.0.1:14552` the error does not happen. Presumably mavsdk creates two connections, one for `127.0.0.1` because that's where messages arrive on, and `localhost` because that's what is created.

I suppose the fix would be to merge the two, and not keep both around. For now, I suggest to use 127.0.0.1 while I look for a fix.

Ba0Nguyen, 2024-11-13:
Ok! Upload mission is working now with using the connection string 'udp://127.0.0.1:14552'. Thank you.

julianoes, 2024-11-14:
No, I just tried and it confirmed my theory.

closed 2024-11-15 by julianoes