mavlink/MAVSDK-Java #183: MAVSDK-JAVA-Client Issue: Deprecated Gradle features used in this build, incompatible with Gradle 9.0

Ghazanfar373, 2024-09-29:
Hi this is Ali, really impressed by your work .Actually I'm creating a javafx gui app to display UAV data, when I tried to implement this library 'io.mavsdk:mavsdk:1.1.1' in my javafx application it gives me this error, I tried different way to resolve but no luck

Error: Deprecated Gradle feature used in this build, making it incompatible with gradle 9.0.

JonasVautherin, 2024-09-29:
How do you include it in your project?

Ghazanfar373, 2024-09-29:
Using Gradle dependency

JonasVautherin, 2024-09-29:
Hmm it should not pull any gradle feature then, should it?

Does it say which deprecated features are used in the build?

Ghazanfar373, 2024-09-30:
> Task :run FAILED
Error occurred during initialization of boot layer
java.lang.module.ResolutionException: Modules grpc.api and grpc.context export package io.grpc to module annotations

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.