mavlink/MAVSDK-Java #184: ArduPilot aircraft low/critical battery action param setting results in SDK erroring out and eventually crashing (v2.1.0)

rayw-dronesense, 2024-09-30:
Setting either `BATT_FS_CRT_ACT` or `BATT_FS_LOW_ACT` with `setParamInt` results in this strange behavior on the SDK where it infinitely loops trying to set, and failing due to "mismatch type", and eventually app crashes with a seg fault.

Example:

```
mavlinkSystem.param.setParamInt("BATT_FS_CRT_ACT", 0).subscribe()
```

```
2024-09-30 13:28:25.201  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 5  (BATT_FS_CRT_ACT).
2024-09-30 13:28:25.207  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.207  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.207  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 4  (BATT_FS_CRT_ACT).
2024-09-30 13:28:25.214  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.214  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.214  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 3  (BATT_FS_CRT_ACT).
2024-09-30 13:28:25.214  4963-5080  Mavsdk                  com.example.mav            W  sending again after 1.52228 s, retries to do: 1  (2504).
2024-09-30 13:28:25.214  4963-5080  Mavsdk                  com.example.mav            W  sending again after 1.52242 s, retries to do: 1  (2505).
2024-09-30 13:28:25.219  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.219  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.219  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 2  (BATT_FS_CRT_ACT).
2024-09-30 13:28:25.223  4963-5083  Mavsdk                  com.example.mav            W  Received ack for not-existing command: 2504! Ignoring...
2024-09-30 13:28:25.226  4963-5083  Mavsdk                  com.example.mav            W  Received ack for not-existing command: 2505! Ignoring...
2024-09-30 13:28:25.227  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.227  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.227  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 1  (BATT_FS_CRT_ACT).
2024-09-30 13:28:25.233  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.233  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.233  4963-5083  Mavsdk                  com.example.mav            E  Error: Retrying failed set param failed: BATT_FS_CRT_ACT
2024-09-30 13:28:25.237  4963-5080  Mavsdk                  com.example.mav            E  Received timeout without already having work requested
2024-09-30 13:28:25.244  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.244  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.244  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 5  (BATT_FS_LOW_ACT).
2024-09-30 13:28:25.248  4963-5080  Mavsdk                  com.example.mav            W  sending again, retries to do: 4  (BATT_FS_LOW_ACT).
2024-09-30 13:28:25.250  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.250  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.250  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 3  (BATT_FS_LOW_ACT).
2024-09-30 13:28:25.256  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.256  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
2024-09-30 13:28:25.256  4963-5083  Mavsdk                  com.example.mav            W  sending again, retries to do: 2  (BATT_FS_LOW_ACT).
2024-09-30 13:28:25.259  4963-5080  Mavsdk                  com.example.mav            W  sending again, retries to do: 1  (BATT_FS_LOW_ACT).
2024-09-30 13:28:25.262  4963-5083  Mavsdk                  com.example.mav            W  Comparison type mismatch between int8_t and float
2024-09-30 13:28:25.262  4963-5083  Mavsdk                  com.example.mav            W  Trying to compare different types.
```

julianoes, 2024-09-30:
Thanks for the issue. Is this against ArduPilot? And which version? And could it be resolved by https://github.com/mavlink/MAVSDK/pull/2401?

rayw-dronesense, 2024-10-02:
Hi @julianoes  - I was able to borrow an x86-64 machine to build the latest MAVSDK C++ into an `.so` and test it out (the source tagged v2.12.7), and can confirm that the linked issue does resolve this issue as well as https://github.com/mavlink/MAVSDK-Java/issues/181

julianoes, 2024-10-03:
Ok thanks for the note. So what you need is an updated Java release? @JonasVautherin any chance you could help with that?

JonasVautherin, 2024-10-04:
I just published a new version that points to mavsdk_server v2.12.6

rayw-dronesense, 2024-10-04:
Can confirm; this issue is fixed with the 2.1.2 release. Thanks @JonasVautherin and @julianoes !

closed 2024-10-04 by rayw-dronesense