mavlink/MAVSDK-Java #186: How to get wind speed or wind direction for IF (inspire flight drone)

gongelosP, 2024-10-02:
Currently working with latest version of this SDK for Android application, already work on the drone connection and getting some information from the telemetry module.
But i can't find a way to obtain the wind speed or wind direction from the telemetry pack, any idea how this is done?
I found in the MAVLINK documentation that there is some wind information https://mavlink.io/en/messages/common.html#WIND_COV

So the question would be how can i get that wind information with this Android SDK

Thanks

julianoes, 2024-10-03:
This message is currently not exposed. We either have to add it to telemetry by adding it to proto first, then C++, and eventually Java.

Also see: https://mavsdk.mavlink.io/main/en/cpp/contributing/autogen.html

In C++ it's possible to subscribe to any MAVLink message using MavlinkPassthrough, however, not in the language wrappers like Java, yet. I'm planning to add (runtime) passthrough for the language wrappers based on https://github.com/Auterion/libmav for v4.

gongelosP, 2024-10-03:
Thanks for the answer @julianoes! First question would be if have any date in mind when you think the MavlinkPassthrough would be available in Java. 

In other hand do you know any other solution to get this wind data in the mean while? Working with Android application and IF drone

julianoes, 2024-10-04:
I can't promise any dates as I do this in my spare time, for free.

If you want to bump this up on my todo list, consider [sponsoring me](https://github.com/sponsors/julianoes).