mavlink/MAVSDK-Java #190: Gimbal Commands Getting Stuck With MAVSDK-server v2.1.3 (V2.12.9 MAVSDK) (It worked correctly on v1.14.6)

rayw-dronesense, 2024-11-20:
## Environment

- PX4-based real aircraft
- MAVSDK-Java 2.1.3 and MAVSDK-Server (v2.12.9 MAVSDK internally)

## Repro

Same exact code as before on real physical aircraft,

```
system.gimbal.setPitchAndYaw(currentAttitude.pitch + 20.0f, 0.0f)
```

which executes when user presses down on a gimbal tilt stick. On 1.3.2 this was working very smoothly.

**Repro video:**

https://github.com/user-attachments/assets/9f462f84-5f48-430c-a904-743efefcfb01

I did notice this log show up in the logcat output, not sure if it has to do with anything,
```
 Falling back to Gimbal Version 1
```

julianoes, 2024-11-20:
What gimbal is this? How is it connected/set up with PX4? Is it using the gimbal version 2 protocol?

You can also try to force it to use gimbal v2 by setting the environment variable `MAVSDK_FORCE_GIMBAL_V2=1`.

rayw-dronesense, 2024-11-20:
Thanks for the suggestions. I tried forcing V2 gimbal with the environment variable:

> Forcing gimbal version 2

Unfortunately, that just made it not work at all.

FWIW, here's the MAVSDK output,

```
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            I  New system on: 192.168.20.30:39722 (with system ID: 1)
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  New system ID: 1 Comp ID: 1
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Autopilot (1) added.
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Companion Computer (191) added.
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Unsupported component (197) added.
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  New system ID: 43 Comp ID: 197
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Unsupported component (197) added.
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Unsupported component (158) added.
2024-11-20 15:53:29.431 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Gimbal (154) added.
2024-11-20 15:53:29.503 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Camera 1 (100) added.
2024-11-20 15:53:29.503 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Discovered 6 component(s)
2024-11-20 15:53:29.503 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  Component Camera 2 (101) added.
2024-11-20 15:53:29.507 11126-11184 Mavsdk                  com.dronesense.pilot.blue            I  System discovered
2024-11-20 15:53:29.510 11126-11184 Mavsdk                  com.dronesense.pilot.blue            I  Server started
2024-11-20 15:53:29.510 11126-11184 Mavsdk                  com.dronesense.pilot.blue            I  Server set to listen on 0.0.0.0:44307
2024-11-20 15:53:29.510 11126-11184 MAVSDK-Server           com.dronesense.pilot.blue            D  mavsdk_server is now running, listening on port 44307
2024-11-20 15:53:29.602 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Vehicle type changed (new type: 2, old type: 0)
2024-11-20 15:53:30.841 11126-11383 Mavsdk                  com.dronesense.pilot.blue            D  MAVLink: warning: Manual control regained
2024-11-20 15:53:31.761 11126-11596 Mavsdk                  com.dronesense.pilot.blue            W  System status is usually fixed at 1 Hz
2024-11-20 15:53:36.877 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Received ack for not-existing command: 527! Ignoring...
2024-11-20 15:53:36.878 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Received ack for not-existing command: 521! Ignoring...
2024-11-20 15:53:36.878 11126-11767 Mavsdk                  com.dronesense.pilot.blue            I  Download file: http://192.168.20.30:8080/greenseer_ir_parameters.xml using cURL...
2024-11-20 15:53:36.878 11126-11767 Mavsdk                  com.dronesense.pilot.blue            I  Downloading camera definition from: http://192.168.20.30:8080/greenseer_ir_parameters.xml
2024-11-20 15:53:36.891 11126-11767 Mavsdk                  com.dronesense.pilot.blue            I  Downloaded file, result Success
2024-11-20 15:53:36.891 11126-11767 Mavsdk                  com.dronesense.pilot.blue            D  Successfully loaded camera definition
2024-11-20 15:53:36.920 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Received ack for not-existing command: 2502! Ignoring...
2024-11-20 15:53:36.920 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Received ack for not-existing command: 527! Ignoring...
2024-11-20 15:53:37.023 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Got unexpected response on work item
2024-11-20 15:53:37.023 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Got unexpected response on work item
2024-11-20 15:53:37.023 11126-11383 Mavsdk                  com.dronesense.pilot.blue            W  Received ack for not-existing command: 527! Ignoring...
```

Actual gimbal looks like this,

[image]

Not sure if that helps

julianoes, 2024-11-20:
Ok, do you know whether you control the gimbal directly, or through PX4? So what are the PX4 `MNT_*` parameters set? What is the output of `gimbal status` in PX4?

rayw-dronesense, 2024-11-21:
Is this the place to check "gimbal status"?

[image]

julianoes, 2024-11-21:
This is how you'd check it. It looks like you're on a quite old PX4 version. In that case the command is `mount status` instead of `gimbal status.

In any case, it looks like it's definitely v1, so I wonder what changed from MAVSDK v1.4 to v2.12. Looking at the diff, I don't see any changes that would touch gimbal:
https://github.com/mavlink/MAVSDK/compare/v2.12...v1.4
I'm not sure what is going on.

Also a note, with MAVSDK v3, gimbal protocol v1 is removed. This will require you to update to a PX4 version which supports gimbal v2. You might be able to try it with your current version but likely it won't work or work well, if it is implemented at all.

rayw-dronesense, 2024-11-22:
@julianoes  Thanks for getting back to me.

Here's what happens when `mount status` is run,

```
nsh> mount status
nsh: mount: missing required argument(s)
nsh> 
```

> Also a note, with MAVSDK v3, gimbal protocol v1 is removed. This will require you to update to a PX4 version which supports gimbal v2. You might be able to try it with your current version but likely it won't work or work well, if it is implemented at all.

This is great to know. However, this is a retail production aircraft from a vendor that we have no control over...

julianoes, 2024-11-25:
Hm, ok, I don't remember the arguments of `mount`. Just do `mount` and it might tell you the options.

> This is great to know. However, this is a retail production aircraft from a vendor that we have no control over...

I'm sorry, me neither :disappointed:. The backwards compatibility stuff was hard to maintain, so I'm afraid it was not worth the trouble for me. If you want to bring it back in and maintain it for v3, let me know. I do MAVSDK in my spare time, so I can't maintain every variation forever.

rayw-dronesense, 2024-11-25:
Workaround found for this issue: wait until gimbal command completes before sending another.

closed 2024-11-25 by rayw-dronesense

julianoes, 2024-11-25:
Aha! Interesting. How do you wait? Do you wait for the result of the MAVSDK function to return? In which case, yes you should always do that.

rayw-dronesense, 2024-11-25:
Very clunky workaround, but works nevertheless,

```
gimbalStickY.take(1).flatMapCompletable { system.gimbal.setPitchAndYaw(...) }.repeat() .subscribe()
```

instead of,

```
gimbalStickY.flatMapCompletable { system.gimbal.setPitchAndYaw(...) }.subscribe()
```