mavlink/MAVSDK-Java #193: Unable to control the drone with  MavSDK-Java Manual Control Plugin

Ba0Nguyen, 2024-12-03:
Ardupilot v4.6 (latest SITL)
MavSDK 2.0.1
SITL starting command: sim_vehicle.py -v Copter --console --map --out=IP:14540

`val completable = mavSdkDataWrapper.manualControlWrapper.move(x, y, z, r)
                completable?.subscribe({
                    Log.d("MotionEvent", "Move Success")
                },{ e ->
                    Log.d("MotionEvent", "Error move: ${e.message}")
                })
`

I'm creating mavsdk server with systemId 245 and compId 190
mavSdkServer.run("IP:14540", 0, 245, 190)

I read another thread that it is solved with systemID 245, I tried that but it still not working and I am not able to control my drone in SITL.

Ba0Nguyen, 2024-12-03:
I was able to receive success result but the drone is still not moving.

julianoes, 2024-12-03:
Regarding the system ID, you can also change the sysid ArduPilot listens on using a parameter. I'm not sure why it wouldn't move though.

Ba0Nguyen, 2024-12-03:
Do you think SITL might have the wrong set up where it is not receiving manual control commands from my joysticks?

Ba0Nguyen, 2024-12-03:
@julianoes I think the problem is how I start SITL, I just need to add the following "output add IP:PORT" to the command line, set the SYSID_MYGCS to the correct one and it is working no\w

closed 2024-12-03 by Ba0Nguyen