mavlink/MAVSDK-Java #194: How to cancel current gimbal command and immediately execute new gimbal command? (v.2.13 with V1 Gimbals)

rayw-dronesense, 2024-12-09:
This is sort of related to a previous issue that I closed with respect to gimbal commands getting stuck and having to wait for the command to complete before issuing another one.

It has come to my attention however, that that particular solution won't be good enough for customers as the expectation is for them to be able to immediately issue a new command expecting to cancel the previous one.

How do we make it so that `gimbal.setangle` or `gimbal.setPitchYaw` can be executed as quickly as desired, and cancels the previous request and executes the new request immediately?

This behavior is observed on two different aircrafts, both using V1 gimbal protocol (one of them is the Gremsy Vio)

JonasVautherin, 2024-12-09:
I have no idea, hopefully @julianoes can help? 🙈

julianoes, 2024-12-11:
There should be two APIs, one to send an angle command with ack. And one that just streams it and uses the MAVLink message (instead of command) under the hood.

And I remember having done this for v3: https://github.com/mavlink/MAVSDK/pull/2363

Coming soon...

closed 2024-12-11 by julianoes

rayw-dronesense, 2024-12-11:
Thanks @julianoes  - will that work with the aircrafts that use "V1 gimbal protocol"?

julianoes, 2024-12-12:
Good question. In theory it could as PX4 should translate it. In practice it might not behave as expected. Needs trying out.

rayw-dronesense, 2024-12-12:
Sure, which APIs would you recommend we try out aside from setPitchYaw and setAngles? https://github.com/mavlink/MAVSDK-Proto/blob/736bc2af1b6eae5c255b5b68014597527bd78c8c/protos/gimbal/gimbal.proto#L18

Both of them have the same issue as reported.

julianoes, 2024-12-12:
The same ones, but with `StreamMode` set to `Stream`:
https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/gimbal/gimbal.proto#L246

rayw-dronesense, 2024-12-12:
@julianoes - Hmm...that seems to be on the "main" branch on the protos repo but not the commit referenced by MAVSDK-Java 2.1.0? https://github.com/mavlink/MAVSDK-Java/tree/main/sdk

There is no `StreamMode` here: https://github.com/mavlink/MAVSDK-Proto/blob/736bc2af1b6eae5c255b5b68014597527bd78c8c/protos/gimbal/gimbal.proto#L18

julianoes, 2024-12-12:
Yes, that will trickle down slowly when v3 comes around.