mavlink/MAVSDK-Java #195: Question: Forwarding RTCM Data with Rtk Plugin

Ba0Nguyen, 2024-12-13:
`            // Read GNSS correction data
            val inputStream = socket.getInputStream()
            val buffer = ByteArray(512)
            var bytesRead: Int
            while (inputStream.read(buffer).also { bytesRead = it } != -1) {
                // Output the binary data
                CoroutineScope(Dispatchers.IO).launch {
                    val rtcmEncodedString: String = android.util.Base64.encodeToString(buffer, android.util.Base64.NO_WRAP)
                    val rtkData = RtcmData(rtcmEncodedString)
                    val completable = rtk.sendRtcmData(rtkData)
                    completable.subscribe({
                        Log.d("RTCM DATA Forwarding", "Forwarding Success, ${rtkData.data}")
                    }, { e ->
                        Log.d("RTCM DATA Forwarding", "Forwarding Failed: error ${e.message}")
                    })
                }
            }`
            
 Is this the correct way of forwarding RTCM Data to the drone to achieve RTK Fixed?
 I got forwarding data result Success but my horizontal uncertainty and vertical uncertainty from the RawGps plugin is still very high. If I leave my FlightController in one place and not move it for 5 minutes, the position uncertainty could get down to less than 3 cm and the flight controller can achieve RTK Fixed status. My goal is to achieve RTK Fixed status at all time when  RTCM data is forwarded. Do you think the way I'm forwarding RTCM Data might affect the position uncertainty? Or could it be that the data I'm getting is not very effective?
 
 Thank you.

julianoes, 2024-12-15:
This should be fixed with v3 where base64 is used.

Ba0Nguyen, 2025-01-13:
How can I update to v3, I don't see Mavsdk-JAVA released v3 in the main project page, is this coming out soon? Thanks!

julianoes, 2025-01-14:
It is coming soon, yes.

Ba0Nguyen, 2025-01-14:
@julianoes Thanks! Do you have an estimated release date for v3, is there any alternative I can do in the mean time to improve my uncertainty and accuracy?

julianoes, 2025-01-14:
What about this week.

Ba0Nguyen, 2025-01-14:
That would be perfect, thank you.

Ba0Nguyen, 2025-02-06:
Hi @julianoes, I'm assuming the maven version MavSDK-Java v2.1.0 is still working with MavSDK v2. I am working in the Android environment, I guess I will need to build mavsdk-server with dockross at the moment to use MavSDK v3 on Android?

julianoes, 2025-02-06:
I'd say v2 should keep working yes.

We'll have to do some work to update everything on v3. PRs welcome.