mavlink/MAVSDK-Java #196: Gremsy Vio with SDK 2.1.5 'Unknown parameter to get option' and 'not found.' when setting GB_MODE which depends on SETTING_TARGET

rayw-dronesense, 2024-12-17:
## Environment

- Sensor: [Gresmy Vio](https://docs.gremsy.com/payloads/vio/camera-setting-menu/camera-settings-menu)
- MAVSDK version 2.14.0 (fork with HTTPS fix)
- MAVSDK-Java version 2.1.5

### Camera definition contents:

https://github.com/Gremsy/Vio-Camera-Definition/releases/download/v2.0.3/vio_camera_f1_def.xml

```
        <parameter name="SETTING_TARGET" type="uint32" default="0">
            <description>Setting Target</description>
            <options>
                <option name="Camera Device" value="0">
                    <exclusions>
                        <exclude>GB_MODE</exclude>
                    </exclusions>
                </option>
                <option name="Gimbal Device" value="1">
                </option>
            </options>
        </parameter>
        <!-- Gimbal params -->
        <parameter name="GB_MODE" type="uint32" default="0">
            <description>Gimbal Mode</description>
            <options>
                <option name="OFF" value="0" />
                <option name="LOCK" value="1" />
                <option name="FOLLOW" value="2" />
                <option name="MAPPING" value="3" />
                <option name="RETURN HOME" value="4" />
            </options>
        </parameter>
```

## Problem

When using `camera.setSetting` on `GB_MODE` this happens,

```
2024-12-17 12:45:09.478 13990-14420 Mavsdk                  com.dronesense.pilot.blue            W  Setting GB_MODE not found.
2024-12-17 12:45:09.478 13990-14420 Mavsdk                  com.dronesense.pilot.blue            E  Unknown parameter to get option: GB_MODE
2024-12-17 12:45:09.478 13990-14420 Mavsdk                  com.dronesense.pilot.blue            E  Could not get option value.
```

It worked okay with `SETTING_TARGET` setting it to "1" first.

julianoes, 2024-12-18:
It's probably best to debug this after the big camera PR has been merged with v3.

rayw-dronesense, 2024-12-18:
Think I found the root cause of the issue, the function `bool CurlWrapper::download_text(const std::string& url, std::string& content)` is only loading the XML file partially, so the parser doesn't see the rest of the file (and thus the missing parameter).

julianoes, 2024-12-18:
Could this be a bug inside of the camera software?

rayw-dronesense, 2024-12-18:
> Could this be a bug inside of the camera software?

Don't think so - QGC works.

I'm having it print out all of the parameters that the `auto e_parameters = e_mavlinkcamera->FirstChildElement("parameters");` finds,

```
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM CAM_MODE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM TRACK_MODE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM OSD_MODE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM RC_MODE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM SETTING_TARGET
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_SOURCE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_REC
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_T_PALETTE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_T_ZOOM
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_ZM_MODE
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_ZM_CB_LV
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_ZM_SR_LV
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_FLIP
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_DEFOG
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_DEFOG_LV
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_AE
2024-12-18 15:44:49.304 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR FOUND PARAM C_V_SP
```

where it stops at "SP"

but the actual XML file has a couple more things under "SP",

```
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:         <parameter name="C_V_AE" type="uint32" default="0">
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:             <description>EO Auto Exposure Mode</description>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:             <options>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                 <option name="Full Auto" value="0">
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                     <exclusions>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_SP</exclude>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_IrP</exclude>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_BrP</exclude>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                     </exclusions>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                 </option>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                 <option name="Manual" value="3">
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                     <exclusions>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_BrP</exclude>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                     </exclusions>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                 </option>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                 <option name="Shutter Priority" value="10">
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                     <exclusions>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_IrP</exclude>
2024-12-18 15:44:49.297 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:                         <exclude>C_V_BrP</exclude>
[... cut ...]
2024-12-18 15:44:49.302 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:             </options>
2024-12-18 15:44:49.302 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:         </parameter>
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE:     </parameters>
2024-12-18 15:44:49.303 18900-19193 Mavsdk                  com.dronesense.pilot.blue            E  RAWR: LINE: </mavlinkcamera>
```

julianoes, 2024-12-18:
Oh but that would have been easier for me :grimacing:

rayw-dronesense, 2024-12-18:
Got it, it's because there are two parameters in the XML that have default values that don't match one of the listed options and the SDK logic bails out as a result,

```
            if (auto default_option = get_default_opt()) {
                new_parameter->default_option = *default_option;
            } else {
                LogErr() << "RAWR BROKE EARLY HERE - the param didnt have valid default value " << param_name;
                return false; // < ---------------- this line here
            }
```

The offending parameters are,

```
        <parameter name="C_V_SP" type="uint32" default="0">
            <description>EO Shutter Value</description>
            <options>
                <option name="1/10" value="13" />
                <option name="1/20" value="14" />
                <option name="1/50" value="17" />
                <option name="1/100" value="20" />
                <option name="1/125" value="21" />
                <option name="1/500" value="25" />
                <option name="1/725" value="26" />
                <option name="1/1000" value="27" />
                <option name="1/1500" value="28" />
                <option name="1/2000" value="29" />
            </options>
        </parameter>
        <parameter name="C_V_IrP" type="uint32" default="0">
            <description>EO Aperture Value</description>
            <options>
                <option name="F2.0" value="25" />
                <option name="F2.2" value="24" />
                <option name="F2.4" value="23" />
                <option name="F2.6" value="22" />
                <option name="F2.8" value="21" />
                <option name="F3.1" value="20" />
                <option name="F3.4" value="19" />
                <option name="F4.0" value="17"/>
                <option name="F5.2" value="14" />
                <option name="F6.8" value="11" />
                <option name="F7.3" value="10" />
                <option name="F8.7" value="8" />
                <option name="F9.6" value="7" />
                <option name="F10" value="6" />
                <option name="F11" value="5" />
            </options>
        </parameter>
```

Looks like QGC just skips over them and continues processing the rest of the config, that's why QGC works...

julianoes, 2024-12-19:
Well, to be fair, that's pretty wrong on their side.

julianoes, 2024-12-19:
@rayw-dronesense any chance you can update the camera?

It looks like this is fixed with:
https://github.com/Gremsy/Vio-Camera-Definition/releases/tag/v8.0.0

rayw-dronesense, 2024-12-19:
Thanks for the suggestions. It doesn't seem like I can update the camera unfortunately as it came packaged with the aircraft in my possession. However, it does look like I can intercept the config downloaded and make the change to the two parameters at least.

That said, for the long run, do we want to match QGC behavior in the SDK, or whichever one is 'correct'?

julianoes, 2024-12-20:
Can you tell your aircraft manufacturer to please update this? That way the ball starts rolling, hopefully.

In terms of behavior, I suppose we can be pragmatic and change it, and just accept broken stuff. Feel free to make a pull request, thanks.