mavlink/MAVSDK-Java #197: Having Issue starting mission

Ba0Nguyen, 2025-01-03:
Sometime I have problem where I can't start a mission even thought a mission is uploaded. Error message gives me an UNKNOWN reason to why it failed to start the mission. If I keep sending start mission command, eventually  MavSDK will start the mission. This might take somewhere around 5-10 tries to get the mission to start.

Ba0Nguyen, 2025-01-03:
I run this with Ardupilot 4.4.4 and MavSDK v2.0.1, should I update to a newer version of MavSDK?

JonasVautherin, 2025-01-03:
Hmm that's weird. [Starting a mission](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L804) is "just" about changing the mode, which is done [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L755).

For Ardupilot, the conversion is done [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L782). See how it calls a different function depending on the vehicle: `flight_mode_to_ardupilot_rover_mode`, `flight_mode_to_ardupilot_plane_mode`, `flight_mode_to_ardupilot_copter_mode`.

I see that `FlightMode::Mission` is defined for [copter](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L886) and [plane](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L919) and [rover](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L857), so I don't understand why it would return `Unknown` 🤔.

julianoes, 2025-01-05:
> MavSDK v2.0.1, should I update to a newer version of MavSDK?

Yes, you should. That's the first step to try.