mavlink/MAVSDK-Java #200: Build not working - missing openssl?

matthieu-db, 2025-03-07:
I'm trying to build the .so in debug mode to debug a crash I get.

I'm following the instructions to build it using dockcross (from the README), but it fails at building curl - openssl seems to be not found?
 
```
CMake Error at /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /work/cmake/Modules/non-msvc/FindOpenSSL.cmake:672 (find_package_handle_standard_args)
  CMakeLists.txt:581 (find_package)
```

I tried compiling openssl with the android toolchain and using it in the build, manually setting the env variables, ...

JonasVautherin, 2025-03-07:
The error message you show says that it doesn't find OpenSSL indeed 👍. However, I can't say much more from this information.

Can you show the commands you are running? The full log output? Feels like maybe OpenSSL failed to build in the earlier step?

matthieu-db, 2025-03-07:
I'm doing the steps in the bottom of the README, actually I just noticed I skipped the
`tools/generate_mavlink_headers.sh` step because that file was not found. Do I need to checkout a particular version of MAVSDK?

matthieu-db, 2025-03-07:
I found out that this generate headers script was no longer a thing, and that the cmake configuration handles that now. I will include the output of cmake . (from inside `./dockcross-android-arm64 bash`) here: 

[output_cmake.txt](https://github.com/user-attachments/files/19130496/output_cmake.txt)

matthieu-db, 2025-03-07:
It doesn't seem there's much I can be doing wrong, considering all the big stuff is happening inside the container

JonasVautherin, 2025-03-07:
Ok, so you run:

> podman run --rm dockcross/android-arm64 > ./dockcross-android-arm64 && chmod +x ./dockcross-android-arm64
./dockcross-android-arm64 cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=OFF -Bbuild/android-arm64 -DMAVLINK_HEADERS="mavlink-headers/include" -H. -DCMAKE_INSTALL_PREFIX=build/android-arm64/install

May I ask what you clone? On which branch/tag are you?

matthieu-db, 2025-03-07:
main branch

I switched to v2.14.1 tag and it seems to be doing better now (still building stuff)

matthieu-db, 2025-03-07:
Nope

```
CMake Error at /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.31/Modules/FindOpenSSL.cmake:691 (find_package_handle_standard_args)
  CMakeLists.txt:430 (find_package)

-- Configuring incomplete, errors occurred!
gmake[2]: *** [CMakeFiles/curl.dir/build.make:92: curl/src/curl-stamp/curl-configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/curl.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  /work/build/android-arm64/third_party/curl failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:23 (build_target)

```

matthieu-db, 2025-03-07:
[out.txt](https://github.com/user-attachments/files/19130776/out.txt)

JonasVautherin, 2025-03-07:
I think it may be related to https://github.com/mavlink/MAVSDK/pull/2518, let me have a look 👍

matthieu-db, 2025-03-07:
I used a dockcross version from before that PR and it has been building for like 15 minutes now so that seems good :)

matthieu-db, 2025-03-07:
It worked! I got a .so! Thanks for the pointer

matthieu-db, 2025-08-22:
Thanks for cleaning up the old issues :)