mavlink/MAVSDK-Java #204: MAV_CMD_DO_SET_MODE (176) .Base on io.dronefleet.mavlink PC JAVA

KKJava1, 2025-06-06:
`COMMAND_LONG {target_system : 1, target_component : 1, command : 176, confirmation : 0, param1 :1 , param2 : 0x30000, param3 : 0.0, param4 : 0, param5 : 0.0, param6 : 0.0, param7 : 0.0}`

目前切换模式是通过这样的参数传递，但是我根据这样的参数进行传递后被拒绝

`
    public void setUavMode(String sn, Integer mode) throws IOException {
        float modeValue = 0;

        UavDataCacheManager cacheManager = UavDataCacheManager.getInstance();

        // 根据传入的模式值设置 modeValue
        switch (mode) {
            case 1:
                // 控制模式
                modeValue = 0x60000;
                break;
            case 2:
                // 定点模式
                modeValue = 0x30000;
                break;
            case 3:
                // 自动模式
                modeValue = 0x4000000;
                break;
            default:
                break;
        }

        // 判断传入的模式和当前模式是否相同
        boolean exists = cacheManager.hasUavData(sn);
        if (exists) {
            Map<String, Object> uavData = cacheManager.getUavDataBySn(sn);
            Object cachedModeNum = uavData.get("modeNum");
            if (cachedModeNum instanceof Float && (Float) cachedModeNum == modeValue) {
                return;
            }
        }

        // 发送命令设置模式
        sendCommand(
                sn,
                MavCmd.MAV_CMD_DO_SET_MODE,
                1,
                Float.parseFloat(Integer.toHexString((int) modeValue)),  // 直接传递十六进制值
                0, 0, 0, 0, 0);

        // 当前线程暂停 1 秒
        long countTime = 1000_000L * 1_000;
        LockSupport.parkNanos(countTime);
    }

    private void sendCommand(
            String sn,
            MavCmd command,
            float param1, float param2, float param3, float param4,
            float param5, float param6, float param7) throws IOException {
        CommandLong cmd = new CommandLong.Builder()
                .command(command)
                .targetSystem(getMavlinkUav(sn).getTargetSystem().getTargetSystem()) // 飞控系统ID
                .targetComponent(getMavlinkUav(sn).getTargetSystem().getTargetComponent()) // 组件ID
                .param1(param1)
                .param2(param2)
                .param3(param3)
                .param4(param4)
                .param5(param5)
                .param6(param6)
                .param7(param7)
                .build();
        MavlinkConnection connection = getMavlinkUav(sn).getMavlinkConnection();
        logger.info("发送的指令为{}",cmd);
        connection.send1(
                properties.getSystemId(),
                properties.getComponentId(),
                cmd);
    }

`

`
2025-06-06 10:23:46.966 - [MQTT Call: re4d3f6a7cc42f0aac8015a1d68913fecd][] INFO  MavlinkService -切换模式的值为393216.0
2025-06-06 10:23:46.967 - [MQTT Call: re4d3f6a7cc42f0aac8015a1d68913fecd][] INFO  .MavlinkService -发送的指令为CommandLong{targetSystem=6, targetComponent=1, command=EnumValue{value=176, entry=MAV_CMD_DO_SET_MODE}, confirmation=0, param1=1.0, param2=393216.0, param3=0.0, param4=0.0, param5=0.0, param6=0.0, param7=0.0}
2025-06-06 10:23:47.010 - [Thread-8][] ERROR com.yuefei.uav.mavlink.service.MavlinkService -指令被拒绝: MAV_RESULT_DENIED，FROM：MAV_CMD_DO_SET_MODE
`
最后报错的结果是这样

JonasVautherin, 2025-06-06:
I am confused. This is the repository of MAVSDK-Java, not io.dronefleet.mavlink, right?

KKJava1, 2025-06-06:
> I am confused. This is the repository of MAVSDK-Java, not io.dronefleet.mavlink, right?

yes，I know，I want to slove my problem，At least they are all implemented in Java and MavLink. At present, I have encountered problems in switching modes. I refer to the code format of MAV-SDK-JAVA here.

```
public void setUavMode(String sn, Integer mode) throws IOException {
    float modeValue = 0;
    UavDataCacheManager cacheManager = UavDataCacheManager.getInstance();
    switch (mode) {
        case 1:
            // Guided mode
            modeValue = 0x60000;
            break;
        case 2:
            // Hold position mode
            modeValue = 0x30000;
            break;
        case 3:
            // Auto mode
            modeValue = 0x4000000;
            break;
        default:
            break;
    }
    boolean exists = cacheManager.hasUavData(sn);
    if (exists) {
        Map<String, Object> uavData = cacheManager.getUavDataBySn(sn);
        Object cachedModeNum = uavData.get("modeNum");
        if (cachedModeNum instanceof Float && (Float) cachedModeNum == modeValue) {
            return;
        }
    }
sendCommand(
        sn,
        MavCmd.MAV_CMD_DO_SET_MODE,
        1,
        modeValue,
        0, 0, 0, 0, 0
    );
}
    private void sendCommand(
            String sn,
            MavCmd command,
            float param1, float param2, float param3, float param4,
            float param5, float param6, float param7) throws Exception {
        CommandLong cmd = new CommandLong.Builder()
                .command(command)
                .targetSystem(getMavlinkUav(sn).getTargetSystem().getTargetSystem()) // 飞控系统ID
                .targetComponent(getMavlinkUav(sn).getTargetSystem().getTargetComponent()) // 组件ID
                .param1(param1)
                .param2(param2)
                .param3(param3)
                .param4(param4)
                .param5(param5)
                .param6(param6)
                .param7(param7)
                .build();
        MavlinkConnection connection = getMavlinkUav(sn).getMavlinkConnection();
        logger.info("发送的指令为{}",cmd);
        connection.send1(
                properties.getSystemId(),
                properties.getComponentId(),
                cmd);
    }
```
I want to change the uav mode,I now want to switch the drone mode, and this mode is also switched according to the document
such as

```
Hold position mode:
COMMAND_LONG {target_system: 1, target_component: 1, command: 176, confirmation: 0, param1: 1, param2: 0x30000, param3: 0.0, param4: 0, param5: 0.0, param6: 0.0, param7: 0.0}
Altitude hold mode:
COMMAND_LONG {target_system: 1, target_component: 1, command: 176, confirmation: 0, param1: 1, param2: 0x20000, param3: 0.0, param4: 0, param5: 0.0, param6: 0.0, param7: 0.0}
Mission mode:
COMMAND_LONG {target_system: 1, target_component: 1, command: 176, confirmation: 0, param1: 1, param2: 0x4000000, param3: 0.0, param4: 0, param5: 0.0, param6: 0.0, param7: 0.0}
Return to home mode:
COMMAND_LONG {target_system: 1, target_component: 1, command: 176, confirmation: 0, param1: 1, param2: 0x5000000, param3: 0.0, param4: 0, param5: 0.0, param6: 0.0, param7: 0.0}
```
At present, the take-off and pointing flight modes have been completed. Are there other points that need to be paid attention to when switching modes, such as some prerequisites and transmission issues, etc.? I look forward to your reply. Thank you.

JonasVautherin, 2025-06-06:
But the code you're showing is not using MAVSDK-Java, is it?