mavlink/MAVSDK-Java #206: NDK Crash "path /tmp is not a directory" when connecting from phone to Mission Planner SITL

AlexShafir, 2025-06-14:
Description: NDK crash happens when I try to connect from Android device to Mission Planner ArduCopter SITL.
Surprisingly, it worked ok some time ago.

Error message: terminating with uncaught exception of type std::__ndk1::__fs::filesystem::filesystem_error: filesystem error: in temp_directory_path: path "/tmp" is not a directory: Not a directory

SITL: Mission Planner, ArduCopter v.4.5.7
Phone: Xiaomi Redmi Note 8T with Android 9
Libs: 
```
implementation("io.mavsdk:mavsdk:3.0.0") {
        exclude(group="com.google.protobuf", module="protobuf-javalite")
}
implementation("io.mavsdk:mavsdk-server:3.0.0")
```

Connect code:
```kotlin
val mMavSdkServer = MavsdkServer()
val systemAddress = "tcp://192.168.178.103:5760"
MavsdkEventQueue.executor().execute {
  val mavsdkServerPort = mMavsdkServer.run(systemAddress)
  val mDrone = System("127.0.0.1", mavsdkServerPort)
}
```
Mavsdk log:
```
MAVSDK version: v3.5.0
Waiting to discover system on tcp://192.168.178.103:5760...
Connection using tcp:// is deprecated, please use tcpin:// or tcpout://
New system ID: 1 Comp ID: 1
TCP receive error: Interrupted system call, trying to reeconnect...
```

NDK Crash:
```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'xiaomi/willow/willow:9/PKQ1.190616.001/V11.0.5.0.PCXMIXM:user/release-keys'
Revision: '0'
ABI: 'arm64'
pid: 29968, tid: 30037, name: mavsdk-event-qu  >>> com.dev.drone <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'terminating with uncaught exception of type std::__ndk1::__fs::filesystem::filesystem_error: filesystem error: in temp_directory_path: path "/tmp" is not a directory: Not a directory'
    x0  0000000000000000  x1  0000000000007555  x2  0000000000000006  x3  0000000000000008
    x4  feff78716e736264  x5  feff78716e736264  x6  feff78716e736264  x7  7f7f7f7f7f7f7f7f
    x8  0000000000000083  x9  2083438910111022  x10 0000000000000000  x11 fffffffc7ffffbdf
    x12 0000000000000001  x13 79726f7463657269  x14 0000000000000008  x15 00000078ee2f2470
    x16 00000079950262b8  x17 0000007994f45ee0  x18 0000000000000010  x19 0000000000007510
    x20 0000000000007555  x21 ffffff80ffffffc8  x22 00000078ee2f2ea0  x23 00000078ee2f2ee0
    x24 00000078ee2f2fc0  x25 00000078ee2f2f08  x26 00000078ee2f4ca0  x27 0020c49ba5e353f7
    x28 0000000000988f2d  x29 00000078ee2f2e10
    sp  00000078ee2f2dd0  lr  0000007994f39354  pc  0000007994f3937c

backtrace:
    #00 pc 000000000002237c  /system/lib64/[libc.so](http://libc.so/) (abort+116)
    #01 pc 00000000000d1cc0  /data/app/com.my.app-rW82WzvYSQQgauARJ7aAWw==/base.apk (offset 0x58000)
```

julianoes, 2025-06-14:
Does this happen in the emulator and/or the real phone?

I suspect it comes from here:
https://github.com/mavlink/MAVSDK/blob/1ff26d163358faa92890eda4452feb1a863896eb/src/mavsdk/core/fs_utils.cpp#L93-L96

AlexShafir, 2025-06-14:
Tested on Pixel 8, API 36 emulator, there I always receive Java MavSdkException: No system.

AlexShafir, 2025-06-15:
Here I attach Xiaomi phone - SITL packets captured in Wireshark.
[phone-sitl-pcap.zip](https://github.com/user-attachments/files/20745571/phone-sitl-pcap.zip)

AlexShafir, 2025-06-15:
Maybe this is related bug https://discuss.ardupilot.org/t/sitl-tcp-socket-closure-when-connecting-with-mavsdk/83806

julianoes, 2025-06-15:
> There I always receive "Java MavSdkException: No system" for setTakeoffAltitude.
MavSdk log:

No system would mean the connection doesn't actually work and once you try to access the system, it crashes, I'd guess.

AlexShafir, 2025-06-16:
On Android Pixel 8 (API 36) I noticed this line in MavSdk log:
`type=1400 audit(0.0:518): avc:  denied  { read } for  name="somaxconn" dev="proc" ino=85190 scontext=u:r:untrusted_app:s0:c215,c256,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0 app=com.my.app`
It means SELinux on Android restricted access to certain api.

Could this be reason for incorrect connection between emulator <-> SITL ?

AlexShafir, 2025-06-18:
I took a look at MavSdk-Java Android sample, broken thing there is expired mapbox key, it should be updated to staart working.
Also, it can be replaced with alternatives that do not require key like vtm-android.
I need some to think what to do best.

I created a small Android app that has following buttons: Connect, Arm, Takeoff, Land and shows status message.
It can be used to reproduce my bug.
Repo here: https://github.com/AlexShafir/MavSdkExample

julianoes, 2025-06-22:
That's great, thanks @AlexShafir. I'll look at this later this week.

julianoes, 2025-06-23:
If you can give me some pointers how you could connect to localhost from within the emulator that might help. So far, I haven't had success. All I see is that the button switches to Disconnect.

julianoes, 2025-06-24:
I'm on Linux... I suppose I should use `--serial0 tcp:0`.

AlexShafir, 2025-06-25:
On Linux I start SITL with these commands (I tested and it is working):

```
cd /home/****/ardupilot
. ~/.profile
sim_vehicle.py -v ArduCopter --no-mavproxy
```

julianoes, 2025-06-25:
Ok, thanks. Are there any steps in terms of connecting from the emulator? Do I have to forward ports?

AlexShafir, 2025-06-25:
> Are there any steps in terms of connecting from the emulator?
For me it worked out-of-the-box. But I tried connecting to SITL from emulator only on Windows.

AlexShafir, 2025-06-30:
I made tests on Linuxb with mavsdk-server 3.0.0
Android 15 emulator - works but takeoff is not working (no action made, there seems some problem with TCP connection)
Android 9 phone - "path /tmp is not a directory" exception.

julianoes, 2025-07-04:
Fix coming: https://github.com/mavlink/MAVSDK/pull/2605

AlexShafir, 2025-07-04:
I observed problem with takeoff is because of SITL magnetometer arm check.

AlexShafir, 2025-07-07:
Could you build artifact so that I can test it?

julianoes, 2025-07-07:
I'm making a release...

AlexShafir, 2025-07-08:
I show status output in UI but I see nothing there.

AlexShafir, 2025-07-08:
Hmm, now takeoff works. Probably this is SITL-related problem.