mavlink/MAVSDK-Java #214: 16 KiB alignement

matthieu-db, 2025-07-14:
libmavsdk_server.so is not 16 KB aligned and this gives a warning from the linter (and makes it crash on newer emulators/phones)

See https://developer.android.com/guide/practices/page-sizes. I tried to figure out how to change mavsdk_server cmake builds to fix this but am not sure, maybe we'd need to add:
`target_link_options(${CMAKE_PROJECT_NAME} PRIVATE "-Wl,-z,max-page-size=16384")`

JonasVautherin, 2025-07-14:
I think it's because we build with NDK 25. My guess is that it's related to this: https://github.com/dockcross/dockcross/pull/895.

You could:

* Cross-compile MAVSDK-C++ for Android using NDK 28 and try again, or
* Help finishing the Dockcross PR. I believe it's just a matter of updating the hash for Stockfish, but someone needs to try it 😊. You could also try to ping the author of the PR to get more details 👍.

Once the dockcross image is released with the newer NDK, we can make a new release of MAVSDK-Java.

matthieu-db, 2025-08-19:
Cool, thanks :)

JonasVautherin, 2025-08-19:
Could you try version 3.10.0 and see if it helps?

matthieu-db, 2025-08-20:
Yes it's solved, thanks :)

closed 2025-08-20 by matthieu-db