mavlink/MAVSDK-Java #217: Android Crash | `filesystem error: Permission Denied`

ProgrammedForAdventure, 2025-08-04:
The latest version of MAVSDK-Java causes a native crash on Android, the primary error message being:
`filesystem_error: filesystem error: in create_directory: Permission denied [/data/local/tmp/mavsdk-component-metadata-a4dc8fef]`

The issue does not appear on version 3.6.0. I tried granting Android additional permissions, but this did not solve the issue. I am pretty confident the mentioned path (`/data/local/tmp`) is a restricted location on Android, that could only be written to on a rooted device.

```
2025-08-04 13:14:13.991 15460-15753 MAVSDK-Server            D  Running mavsdk_server with connection url: udpout://<drone_ip>:14550
2025-08-04 13:14:13.991 15460-15753 Mavsdk                   I  Waiting to discover system on udpout://<drone_ip>:14550...
2025-08-04 13:15:27.158 15460-15756 Mavsdk                   D  New system ID: 1 Comp ID: 1
2025-08-04 13:15:27.162 15460-15756 libc++abi                E  terminating with uncaught exception of type std::__ndk1::__fs::filesystem::filesystem_error: filesystem error: in create_directory: Permission denied [/data/local/tmp/mavsdk-component-metadata-a4dc8fef]
2025-08-04 13:15:27.186 15460-15756 libc                     A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 15756 (mavsdk-event-qu), pid 15460 (myapp)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A  Cmdline: myapp
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A  pid: 15460, tid: 15756, name: mavsdk-event-qu  >>> myapp <<<
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #01 pc 00000000000d1fcc  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #02 pc 00000000000d2180  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #03 pc 00000000000e6e80  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #04 pc 00000000000e6514  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #05 pc 00000000000e646c  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (__cxa_throw+124) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #06 pc 000000000007cbb4  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #07 pc 000000000007fc54  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #08 pc 00000000000803f0  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libc++_shared.so (offset 0x20c8000) (std::__ndk1::__fs::filesystem::__create_directory(std::__ndk1::__fs::filesystem::path const&, std::__ndk1::error_code*)+140) (BuildId: df3f83af4e54f316c793c3655c4753a8851a234c)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #09 pc 00000000017f9224  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::create_tmp_directory(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&)+656)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #10 pc 00000000017e56c8  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::MavlinkComponentMetadata::MavlinkComponentMetadata(mavsdk::SystemImpl&)+1144)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #11 pc 00000000017bff5c  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::SystemImpl::SystemImpl(mavsdk::MavsdkImpl&)+612)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #12 pc 00000000017bfb84  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::System::System(mavsdk::MavsdkImpl&)+68)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #13 pc 00000000017d285c  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::MavsdkImpl::make_system_with_component(unsigned char, unsigned char)+632)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #14 pc 00000000017d19b0  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::MavsdkImpl::process_message(__mavlink_message&, mavsdk::Connection*)+2688)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #15 pc 00000000017d0ec4  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::MavsdkImpl::process_messages()+112)
2025-08-04 13:15:27.432 15899-15899 DEBUG                    A        #16 pc 00000000017cef4c  /data/app/~~XODcxK6cmAIVh-OPmfQN-g==/myapp-YIgzs15ssLvXR7zzlCwbYg==/base.apk!libmavsdk_server.so (offset 0xaf38000) (mavsdk::MavsdkImpl::work_thread()+108)
```

JonasVautherin, 2025-08-04:
Yes, `/data/local/tmp` will probably never work on modern Android. So you should probably set the path manually.

I am not sure where this is used: are you saying that you don't need it at all and shouldn't have to set the path manually? Because I guess it wasn't writing a file before 3.6.0 anyway 🤔.

@julianoes: is this a path that could be made optional instead of hardcoded to `/data/local/tmp`?

ProgrammedForAdventure, 2025-08-04:
I see there is a new `setTempDirectory()` command like you mentioned -- is there something I should throw in here for now as a workaround? I didn't know it existed.

matthieu-db, 2025-08-05:
Same behavior here, I fixed it with mavsdkServer.setTempDirectory(applicationContext.cacheDir.absolutePath)

JonasVautherin, 2025-08-20:
I think it's not ideal to crash by default. Trying to improve it [here](https://github.com/mavlink/MAVSDK-Java/pull/220).

Would someone be willing to try it? I am publishing `3.10.0-3-SNAPSHOT` if someone wants to test (see the docs to use SNAPSHOTs [here](https://github.com/mavlink/MAVSDK-Java?tab=readme-ov-file#snapshots)).

PR #220 (2025-08-21): Set a default temporary folder for MavsdkServer
We either need a context to create a temporary dir, or ask the user to give us a path. 

Resolves #217.
closed 2025-08-21 by JonasVautherin