mavlink/MAVSDK-Java #218: java.lang.UnsatisfiedLinkError on v3.10.0

akirahrkw, 2025-08-20:
Thank you for the release of v3.10.0
https://github.com/mavlink/MAVSDK-Java/releases/tag/3.10.0

I encountered the following exception when testing v3.10.0 on my Android app, which does not occur in v3.9.0:

```

java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZTVNSt6__ndk114basic_ofstreamIcNS_11char_traitsIcEEEE" referenced by "/data/app/~~-H8LEZTA9G6jJOAoq9oLAA==/com.test.mavaskapp-q8ILAylme6IY5Fu7GLnMtg==/lib/arm64/libmavsdk_server.so"...
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:1090)
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:1012)
                                                                                                    	at java.lang.System.loadLibrary(System.java:1765)
                                                                                                    	at io.mavsdk.mavsdkserver.MavsdkServer.<clinit>(MavsdkServer.java:6)
```

**Environment:**
- MAVSDK-Java version: v3.10.0 (working in v3.9.0)
- Platform: Android

I'm not sure if this is something I'm missing for configuration, or is this a potential regression?

julianoes, 2025-08-20:
Looks like a regression, oops.

JonasVautherin, 2025-08-20:
@akirahrkw: when does it happen? Directly when creating the `MavsdkServer` object?

@matthieu-db: you seem to have had success [here](https://github.com/mavlink/MAVSDK-Java/issues/214). Were you building from sources or using 3.10.0 from Maven Central? Have you seen this UnsatisfiedLinkError?

> Or an ndk issue?

The NDK was updated indeed, also the SDK. I can try to update MAVSDK-Java's minSdk to 24, which will match the one in dockcross. Maybe that will help 🤞

JonasVautherin, 2025-08-20:
MAVSDK-Java v3.10.0-1 will be on Maven Central in a few minutes. @akirahrkw would you mind trying it and letting me know if it helps?

matthieu-db, 2025-08-20:
I didn't actually run it, I just compiled with the dependency. I will try v3.10.0-1 and also report back

matthieu-db, 2025-08-20:
Same error for me with 3.10.0-1

matthieu-db, 2025-08-20:
> Okay now it seems to work but my app is crashing on some mapbox related code, so hard to test any further for now

When I move back to 3.9.0 it works... but on the snapshot I get:
```
terminating due to uncaught exception of type std::bad_cast: std::bad_cast
Pointer tag for 0x716fbecc50 was truncated, see 'https://source.android.com/devices/tech/debug/tagged-pointers'.
Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 3590 (MB Worker U 1), pid 3459 (.id.debug)
Cmdline: com.app.id.debug
pid: 3459, tid: 3590, name: MB Worker U 1  >>> com.app.id.debug <<<
      #01 pc 00000000000d2a84  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libnative_lib.so (BuildId: 186dd2e09aaa4d91f986c9b975bd11fb5c4dc415)
      #02 pc 00000000000d23a4  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libnative_lib.so (BuildId: 186dd2e09aaa4d91f986c9b975bd11fb5c4dc415)
      #03 pc 00000000000d2264  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libnative_lib.so (BuildId: 186dd2e09aaa4d91f986c9b975bd11fb5c4dc415)
      #04 pc 00000000000eca98  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libnative_lib.so (BuildId: 186dd2e09aaa4d91f986c9b975bd11fb5c4dc415)
      #05 pc 00000000000eca74  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libnative_lib.so (__cxa_throw+88) (BuildId: 186dd2e09aaa4d91f986c9b975bd11fb5c4dc415)
      #06 pc 00000000000eea84  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libc++_shared.so (BuildId: e8503e76ded720095755e58f7c0e4b30cf32df8c)
      #07 pc 0000000000112cdc  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libc++_shared.so (std::__ndk1::locale::use_facet(std::__ndk1::locale::id&) const+132) (BuildId: e8503e76ded720095755e58f7c0e4b30cf32df8c)
      #08 pc 00000000002be148  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libmapbox-common.so (BuildId: 136773315c2cdab930bf485dc846e48962970cf1)
      #09 pc 00000000002d38a8  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libmapbox-common.so (BuildId: 136773315c2cdab930bf485dc846e48962970cf1)
      #10 pc 000000000034b39c  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libmapbox-common.so (mapbox::common::platform::setCurrentThreadServiceType(mapbox::common::ThreadServiceType)+1348) (BuildId: 136773315c2cdab930bf485dc846e48962970cf1)
      #11 pc 000000000032f598  /data/app/~~dXKqKQ-rRJQ44aiAeqNVKg==/com.app.id.debug-xZbh7j5msIgjUNoEKbmK1g==/lib/arm64/libmapbox-common.so (BuildId: 136773315c2cdab930bf485dc846e48962970cf1)
```

matthieu-db, 2025-08-20:
@akirahrkw could you try 3.10.0-2-SNAPSHOT as well? https://github.com/mavlink/MAVSDK-Java/issues/218#issuecomment-3206213959

matthieu-db, 2025-08-20:
OK I think I have some kind of mismatch on the c++ stdlib with mapbox then... Ugh I don't know how to fix that 😢

JonasVautherin, 2025-08-20:
> OK I think I have some kind of mismatch on the c++ stdlib with mapbox then... Ugh I don't know how to fix that 😢

@matthieu-db: What I changed is that I removed `libc++_shared.so` from the mavsdk-server library. It was superfluous because mavsdk_server uses a C API. Actually I think it was bad because it was bringing a `libc++_shared.so` that could conflict with the app.

But maybe the opposite was happening to you: your app was picking the `libc++_shared.so` coming from MAVSDK, and now it's using another one (and that fails for some reason)? 🤔 

Are you using the NDK at all in your project? Or is it only coming through third-parties you import (like mapbox and mavsdk)? If you use the NDK, which version are you using?